tpaviot/pythonocc-core #741: No module named 'OCC' after successfully install

keleisi2001, 2019-12-27:
I used this sentence and successfully installed：
conda create -n pythonocct -c conda-forge -c tpaviot -c 3dhubs -c pythonocc -c dlr-sc -c oce pythonocc-core=7.4.0beta python=3.7

I want to use 2 pictures to say it：

keleisi2001, 2019-12-27:
I found github issue doesn't allow user to upload pics, so i send an email to Paviot.

tpaviot, 2019-12-27:
You've been using idle. Be careful, there are as many idle as environments. Be sure to launch the idle located in the current environment.

keleisi2001, 2019-12-27:
> You've been using idle. Be careful, there are as many idle as environments. Be sure to launch the idle located in the current environment.

Do you mean I have successfully installed it? Can you give me some simple test code? I will test in vs code and pycharm. You know, I have used 4 days to install it. I really really hope it can be done!

keleisi2001, 2019-12-27:
> You've been using idle. Be careful, there are as many idle as environments. Be sure to launch the idle located in the current environment.

I used your latest example in PyCharm, but it was also failed. Screenshot is sent by a new email.

tpaviot, 2019-12-27:
You have to learn how to use conda and switch from an environment to the other, please refer to the related documentation.

closed 2019-12-27 by tpaviot