tpaviot/pythonocc-core #747: How to calculate center of mass of an ellipse

bernhard-42, 2019-12-31:
If I create an ellipse like that

```python
from OCC.Core.gp import gp_Ax1, gp_Ax2, gp_Elips, gp_Pnt, gp_Dir
from OCC.Core.GC import GC_MakeArcOfCircle, GC_MakeArcOfEllipse
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge

pnt = gp_Pnt(0.0, 0.0, 0.0)
dir = gp_Dir(0.0, 0.0, 1.0)

ellipse_gp = gp_Elips(gp_Ax2(pnt, dir), 4000, 40)
ellipse = BRepBuilderAPI_MakeEdge(ellipse_gp).Edge()
```

and try to calculate the center of mass with `brepgprop_LinearProperties` (which I understood is the correct algorithm for `TopoDS_Edge`):

```python
from OCC.Core.GProp import GProp_GProps
from OCC.Core.BRepGProp import brepgprop_LinearProperties

Properties = GProp_GProps()
brepgprop_LinearProperties(ellipse, Properties)
com = Properties.CentreOfMass()
print("ellipse (linear): %9.6f %9.6f %9.6f" % (com.X(), com.Y(), com.Z()))
```
I get:

```python
ellipse (linear): -81.295066  0.000000  0.000000
```

For an ellipse around (0,0,0) I would have expected the center to be (0,0,0).

If I use `brepgprop_SurfaceProperties`

```python
from OCC.Core.GProp import GProp_GProps
from OCC.Core.BRepGProp import brepgprop_SurfaceProperties

Properties = GProp_GProps()
brepgprop_SurfaceProperties(ellipse, Properties)
com = Properties.CentreOfMass()
print("ellipse (surface): %9.6f %9.6f %9.6f" % (com.X(), com.Y(), com.Z()))
```

I get the correct result:

```python
ellipse (surface):  0.000000  0.000000  0.000000
```

What is the correct approach here?

bernhard-42, 2019-12-31:
btw. this behavior is the same for OCC 0.18 and 7.4 rc1

tpaviot, 2019-12-31:
I'll have a look. Is it the same for ```gp_Circ``` ?

bernhard-42, 2019-12-31:
Thanks @tpaviot.
No, for `gp_Circ` all is fine. It also works for `gp_Elips` when `major_radius == minor_radius` (i.e. again a circle). And the larger `major_radius/minor_radius` gets the larger the deviation of the center is.

tpaviot, 2019-12-31:
The first one (linear) is the good one. The second (surface) returns 0 because it's a 1D element whereas the methods expects a 2D. You can test with the ```Mass()``` property of the ```Properties``` variable, it should return the edge length. Works perfectly for a ```gp_Circ```. For a ```gp_Elips```, are you sure the ellips is centered at the origin ?

bernhard-42, 2019-12-31:
I would have expected that this creates a ellipse centered at (0,0,0)
```python
pnt = gp_Pnt(0.0, 0.0, 0.0)
dir = gp_Dir(0.0, 0.0, 1.0)
ellipse_gp = gp_Elips(gp_Ax2(pnt, dir), 4000, 40)
```
If I create the bounding box, I also get (0,0,0) as the center of the bounding box. 
Do I miss something?

tpaviot, 2019-12-31:
did you try with ```gce_MakeElips```

bernhard-42, 2019-12-31:
Same result:

```python
from OCC.Core.gp import gp_Ax1, gp_Ax2, gp_Elips, gp_Pnt, gp_Dir
from OCC.Core.gce import gce_MakeElips
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge

pnt = gp_Pnt(0.0, 0.0, 0.0)
dir = gp_Dir(0.0, 0.0, 1.0)

ellipse_gce = gce_MakeElips(gp_Ax2(pnt, dir), 4000, 40)
ellipse = BRepBuilderAPI_MakeEdge(ellipse_gp).Edge()

Properties = GProp_GProps()
brepgprop_LinearProperties(ellipse, Properties)
com = Properties.CentreOfMass()
print("ellipse (linear): %9.6f %9.6f %9.6f" % (com.X(), com.Y(), com.Z()))

# ellipse (linear): -81.295066  0.000000  0.000000
```

bernhard-42, 2019-12-31:
Going back to gp_Elips. Build an arc from 0 to pi and one from -pi/2 to pi/2 and get the startpoint end endpoint

```python
from OCC.Core.gp import gp_Ax1, gp_Ax2, gp_Elips, gp_Pnt, gp_Dir
from OCC.Core.GC import GC_MakeArcOfCircle, GC_MakeArcOfEllipse
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeFace
from OCC.Core.BRepAdaptor import BRepAdaptor_Curve

pnt = gp_Pnt(0.0, 0.0, 0.0)
dir_ = gp_Dir(0.0, 0.0, 1.0)

ellipse_gp = gp_Elips(gp_Ax2(pnt, dir_), 4000, 40)
ellipse = BRepBuilderAPI_MakeEdge(ellipse_gp).Edge()

import math

ellipse_geom = GC_MakeArcOfEllipse(ellipse_gp, 0.0, math.pi, True).Value()
edge1 = BRepBuilderAPI_MakeEdge(ellipse_geom).Edge()

ellipse_geom = GC_MakeArcOfEllipse(ellipse_gp, math.pi/2, -math.pi/2, True).Value()
edge2 = BRepBuilderAPI_MakeEdge(ellipse_geom).Edge()

def vertices(edge, first=True):
    curve = BRepAdaptor_Curve(edge)
    if first:
        u = curve.FirstParameter()
    else:
        u = curve.LastParameter()
    pnt = curve.Value(u)
    print(("%12.6f "*3) % (pnt.X(), pnt.Y(), pnt.Z()))

vertices(edge1, True), vertices(edge1, False)
vertices(edge2, True), vertices(edge2, False)
```

results in

```python
4000.000000     0.000000     0.000000 
-4000.000000     0.000000     0.000000 
    0.000000    40.000000     0.000000 
   -0.000000   -40.000000     0.000000 
```

So the ellipse is centered and I would expect (0,0,0) as center of mass

rainman110, 2019-12-31:
I suppose this is an error of OCCT. Therefore, it is better to post this issue directly at the OCCT mantis (issue tracker of opencascade). Pythonocc simply makes OCCT available to python users and therefore will contain all glitches of OCCT.

bernhard-42, 2019-12-31:
@rainman110 Happy to do that, however, I have no idea of C++ or opencascade. 
I come from cadquery and the port of my problem from cadquery to pythonocc brought me to the limit of my occ understanding. Can I post pythonocc code there and would they understand python  or am I supposed to port my problem to C++ (which I wouldn't be able).
Is this the URL: https://tracker.dev.opencascade.org/my_view_page.php and do I need to register?

Sorry for the questions, but going to a C++ issue tracker is slightly out of my comfort zone ;-)

bernhard-42, 2020-01-01:
@rainman110 Luckily I found a cpp example of the OCC bottle (with Makefile, puhhh) and I can confirm that by using `gp_Elips` in OCC 7.4.0_1 the center is shifted while the ellipse is perfectly centered

```cc
void TestEllipse() {
  gp_Pnt center(0., 0., 0.);
  gp_Dir zDir = gp::DZ();
  gp_Ax2 axis(center, zDir);
  gp_Elips ellipse_gp(axis, 4000., 40.);
  TopoDS_Edge ellipse = BRepBuilderAPI_MakeEdge(ellipse_gp);
  
  Bnd_Box bb;
  BRepBndLib::AddOptimal(ellipse, bb);
  
  double xmin, ymin, zmin, xmax, ymax, zmax;
  bb.Get(xmin, ymin, zmin, xmax, ymax, zmax);
  
  printf("ellipse (bb x): [%12.6f, %12.6f]\n", xmin, xmax);
  printf("ellipse (bb y): [%12.6f, %12.6f]\n", ymin, ymax);
  printf("ellipse (bb z): [%12.6f, %12.6f]\n", zmin, zmax);

  GProp_GProps Properties;
  BRepGProp::LinearProperties(ellipse, Properties);
  gp_Pnt c = Properties.CentreOfMass();

  printf("ellipse (center): %9.6f %9.6f %9.6f\n", c.X(), c.Y(), c.Z());
}
```

results in 

```
ellipse (bb x): [-4000.000000,  4000.000000]
ellipse (bb y): [  -40.000000,    40.000000]
ellipse (bb z): [    0.000000,     0.000000]
ellipse (center): -81.295066  0.000000  0.000000
```

rainman110, 2020-01-01:
Congrats! Welcome to the C++ devs. 

Instead of posting this full example to the occt Mantis, you could just export the edge as a BREP file.

bernhard-42, 2020-01-01:
I have created an issue at the OCCT mantis: https://tracker.dev.opencascade.org/view.php?id=31290

rainman110, 2020-01-01:
Perfect! This is the best way to make sure that this but will also be fixed in future! 
If you don't mind, you can close this issue.

closed 2020-01-01 by bernhard-42

bernhard-42, 2020-01-01:
For completeness, the answer from OCCT mantis:

>     This seems to be a bug in calculation of linear properties of an ellipse.
> 
>     Workaround is to create a face of that ellipse and to compute surface properties of that face.
> 
>     The computed center of such face is (0, -3.91083e-15, 0).
>

tpaviot, 2020-01-02:
Good catch !