tpaviot/pythonocc-core #748: StlAPI_Reader: No constructor defined

OlegJakushkin, 2020-01-03:
So in latest conda version for python 3.6 on such code:

```Python
from OCC.Display.SimpleGui import init_display
from OCC.Core.TopoDS import TopoDS_Shape
from OCC.Core.StlAPI import StlAPI_Reader

stl_reader =  StlAPI_Reader()
fan_shp = TopoDS_Shape()
test = StlAPI_Reader.Read(fan_shp, stl_file)
```
I get:

> AttributeError                            Traceback (most recent call last)
> <ipython-input-35-897cdf22dc2c> in <module>()
>      29 from OCC.Core.StlAPI import StlAPI_Reader
>      30 
> ---> 31 stl_reader =  StlAPI_Reader()
>      32 fan_shp = TopoDS_Shape()
>      33 test = StlAPI_Reader.Read(fan_shp, stl_file)
> 
> /opt/conda/lib/python3.6/site-packages/OCC/Core/StlAPI.py in __init__(self, *args, **kwargs)
>     167 
>     168     def __init__(self, *args, **kwargs):
> --> 169         raise AttributeError("No constructor defined")
>     170     __repr__ = _swig_repr
>     171     Read = _swig_new_instance_method(_StlAPI.StlAPI_Reader_Read)
> 
> AttributeError: No constructor defined

And one can not simply call `StlAPI_Reader.Read` because it reqieres ` 'StlAPI_Reader_Read', argument 1 of type 'StlAPI_Reader *'`

Intresting point is that STL  Writer works fine:
```Python
stl_exporter = StlAPI_Writer()
stl_exporter.SetASCIIMode(True)  # change to False if you need binary export
stl_exporter.Write(my_torus, stl_file)
```

rainman110, 2020-01-03:
Thanks @OlegJakushkin  for posting this issue.

I think, the issue comes from the ```nodefaultctor``` keyword, see
https://github.com/tpaviot/pythonocc-core/blob/b25f6fc18b6238b1143023cb3af48574b16f0c9f/src/SWIG_files/wrapper/StlAPI.i#L107

@tpaviot Why do you use ```nodefaultctor``` for all classes?

tpaviot, 2020-01-03:
@OlegJakushkin You can use the ```stlapi_Read``` function instead.

@rainman110 The ```%nodefaultctor``` directive was added a long time ago, such a long time that I don't even remember why. But I remember this directive was necessary to compile the library, I should try to remove it to see how it affects classes.

All available constructors are handled by the wrapper. The ```StlAPI_Reader``` class don't have any constructor anymore (have a look at the StlAPI_Reader.hxx header).
```

OlegJakushkin, 2020-01-03:
@tpaviot `stlapi_Read` worked for me!
[image]

For those who will play around with it here is how to build PythonOCC+OCE on conda+ubuntu16 (where I got by the time @tpaviot answered):
```bash
#build PyOCC with OCE 7.4
add-apt-repository -y ppa:timsc/swig-3.0.12 && apt update
apt install -fy git libpcre3-dev cmake bison flex autotools-dev automake libx11-dev mesa-common-dev freeglut3-dev libfreetype6-dev libc6-dev libstdc++6 libstdc++-4.8-dev build-essential gcc g++ libftgl-dev xorg-dev tcl-dev tk-dev python3-dev swig 
wget "https://git.dev.opencascade.org/gitweb/?p=occt.git;a=snapshot;h=fd47711d682be943f0e0a13d1fb54911b0499c31;sf=tgz" && mv index*z occt.tgz && tar -xvzf occt.tgz
cd occt-fd47711 && mkdir cmake-build
cd occt-fd47711 && cd cmake-build  && cmake -D CMAKE_C_FLAGS="-O3 -march=native" .. && make '-j'$(nproc) && make install/strip
#!git clone --branch "7.4.0rc1" git://github.com/tpaviot/pythonocc-core.git
cd pythonocc-core && mkdir cmake-build 
# Note possible path errors with SWIG and special chars # !cp -fr pythonocc-core /pythonocc-core
cd ./pythonocc-core && cd cmake-build && cmake -D CMAKE_C_FLAGS="-O3 -march=native" -DOCE_INCLUDE_PATH=/usr/local/include/opencascade/ -DOCE_LIB_PATH=/usr/local/lib .. && make '-j'$(nproc) && make install
```

tpaviot, 2020-01-03:
ok, I'll close the issue as soon as I tested to remove ```nodefaultctor``` and see what happens

rainman110, 2020-01-03:
> All available constructors are handled by the wrapper. The StlAPI_Reader class don't have any constructor anymore (have a look at the StlAPI_Reader.hxx header).

In C++, default constructors will be generated if not explicitly deleted or any other non-default constructors are defined. By disabling wrapping the default constructors with swig, they won't be available to python though.

tpaviot, 2020-01-04:
@rainman110 I had a look at the pythonocc-generator, there is a function ```class_can_have_default_constructor``` (see https://github.com/tpaviot/pythonocc-generator/blob/master/src/generate_wrapper.py#L1477) that decides whether or not to use the ```%nodefaultctor``` directive. I guess I have to update it.

tpaviot, 2020-01-04:
I confirm that removing ```%nodefaultctor``` works for *almost* all classes. When I'm done with 7.4.0 I will start refactoring completely the generator, there is too much undocumented code and special cases handled in an ugly way.

tpaviot, 2020-01-04:
Fixed by commit 2a7e46f7b1c6cbf52815a8d5f2e14a989f721f7c

tpaviot, 2020-01-05:
Fixed by PR#749

closed 2020-01-05 by tpaviot