tpaviot/pythonocc-core #752: maintain on conda-forge

HalfManHalfChicken, 2020-01-06:
My new year wish is that pythonocc-core can maintain a package on conda-forge:

* Compatibility with a much larger python eco system
* ability to leverage power automatic update tools to avoid hassle when deploying to various architecture
* ensured binary compatibility with other's software

Thank you very much.

tpaviot, 2020-01-06:
I disagree with your first point.

For the second point, pythonocc-core already provides a CI workflow.

At last, regarding the third point, it's not a conda-forge feature, it's a conda drawback.

But I have a good news though: your dream can becomes reality, you just have to take care of it.

looooo, 2020-02-24:
> For the second point, pythonocc-core already provides a CI workflow.

Automatic updating means automatic migration to newer deps which is a really nice feature of conda-forge. Having python-occ as part of conda-forge would be nice, as conda-forge tries to have a consistent occt version via pinning occt for all occt dependent versions (currently 7.4).

> But I have a good news though: your dream can becomes reality, you just have to take care of it.

I am looking forward to do so for quite some time. Just waiting for the 7.4 release.

tpaviot, 2020-02-24:
Great, I just tagged 7.4.0 a couple of days ago. I did not make any announce yet, I wait for conda packages to be ready, but you can start trying to make something with conda-forge right now if you want.

looooo, 2020-02-24:
Thanks, PR added: https://github.com/conda-forge/staged-recipes/pull/10919

Is the opencascade library used in the recipe of this repo patched or why are you not using the library from conda-forge?

tpaviot, 2020-02-24:
I used to work with @rainman110 from dlr-sc to package pythonocc for conda, I never use conda-forge's occt7.4.0, but it should work as is. It should IMO be better to use conda-forge opencascade package to package pythonocc at conda-forge, it would be more consistent.

rainman110, 2020-02-24:
@tpaviot the 7.4.0 release is building right now.

Thanks for notifying me!

looooo, 2020-02-24:
I think the conda-forge recipe is in a mergeable state. Please let me know if  I should add you to the maintainer list @rainman110 @tpaviot 
https://github.com/conda-forge/staged-recipes/pull/10919

tpaviot, 2020-02-24:
Thank you for your contribution @looooo I suggest you're the only maintainer

The linux_64 build build seems to be failing though

tpaviot, 2020-02-24:
@looooo note that you don't need to run the tests to make the package. On my azure dev pipeline, I use to remove run_test* before packaging.

looooo, 2020-02-24:
> The linux_64 build build seems to be failing though

It should be fine after restarting the ci.

> @looooo note that you don't need to run the tests to make the package. On my azure dev pipeline, I use to remove run_test* before packaging.

It's not bad to have some tests in addition to the import-tests.

looooo, 2020-02-25:
py3.7 packages are available [1]. py3.8 should be uploaded soon. We can enable py3.6 if needed, but I guess conda-forge will stop supporting py3.6 once py3.8 is the new default.

Feel free to close this issue.

tpaviot, 2020-02-25:
That's a great contribution, thank you @looooo for this work. py3.6 is not necessary, 3.7 and 3.8 are enough I guess.

closed 2020-02-25 by tpaviot