tpaviot/pythonocc-core #754: TypeError: in method 'new_TopExp_Explorer', argument 1 of type 'TopoDS_Shape const &'

khyatiip, 2020-01-09:
I'm using the following code to obtain vertices using ifcopenshell and pythonocc but i get a type error. I'm attaching a screenshot of the error. I installed ifcopenshell and pythonocc using conda install.

[image]

tpaviot, 2020-01-09:
which version of pythonocc do you use ?

```
import OCC
print(OCC.VERSION)
```

khyatiip, 2020-01-09:
I'm using this version of OCC: 7.4.0-rc1

khyatiip, 2020-01-10:
I got it working. In ifcopenshell/geom/main.py and ifcopenshell/geom/occ_utlis.py wherever `OCC.something` is being used, change it to `OCC.Core.something` 
This may not be the correct/ideal solution but it is working for me as of now.

I don't know if the outputs are correct or not but I'm not this error anymore.

closed 2020-01-10 by khyatiip

rainman110, 2020-01-10:
For me, this sounds as if you have both pythonocc 7.4.0 and pythonocc 0.17 installed. I am not sure, why this is even possible to have both installed.

In pythonocc 7.4.0, you have to import ```OCC.Core```, in older versions, just ```OCC```.

twut, 2020-02-04:
Hi, I have the same issue when I used pythonocc 0.17.3
```python
shape = ifcopenshell.geom.create_shape(settings, ifc_element)
exp = OCC.TopExp.TopExp_Explorer(shape.geometry, OCC.TopAbs.TopAbs_VERTEX)
```
TypeError: in method 'new_TopExp_Explorer', argument 1 of type 'TopoDS_Shape const &'
I tryed the same code in the pythonocc 7.4.0 with OCC.Core, then it works
```python
shape = ifcopenshell.geom.create_shape(settings, ifc_element)
exp = OCC.Core.TopExp.TopExp_Explorer(shape.geometry, OCC.Core.TopAbs.TopAbs_VERTEX)
```
It looks the TopExp_Explorer function take diffierent data types in these 2 different version, TopoDS_Shape,  TopoDS_Shape & 
Is there a way to solve this issue in the pythonocc 0.17.3?

tpaviot, 2020-02-04:
pythonocc 0.17.3 is really old, it's not maintained anymore.