tpaviot/pythonocc-core #755: I am using pythonocc to develop 3D Drawing of Steel Connection. In addition, I would like to develop 2D Drawing for different views from developed

Dar5han7, 2020-01-10:


tpaviot, 2020-01-10:
Can you please provide a description.

Dar5han7, 2020-01-17:
[image]
[image]
[image]
[image]
Attached 3D images and 2D images of the same. Currently, we are creating 2D image by defining the co-ordinate and tracing each and every line. We would like to trace the same from the developed 3D like "flatshot" command in AutoCaD. Is it possible?

Dar5han7, 2020-01-22:
awaiting your reply

tpaviot, 2020-01-22:
I don't understand your question. I don't know Autocad and the "flatshot" command. Please @Dar5han7 contact me by private email if you need further assistance. Otherwise, please use the appropriate vocabulary, and don't put any pressure on anybody on this issue tracker.

Durksz, 2020-02-02:
Dear Dar5han7, 
I think you can better ask this question on the Opencascade forum since your question is related to design using OpenCascade (OCCT), Pythonocc is just a wrapper of OCCT.

looooo, 2020-03-25:
I guess FreeCAD with Techdraw is the closest you can get. I am currently working on something similar (full parametric drawing with annotations of a gear-rack). There are some difficulties of the annotations (jumping around on parameter changes / topo naming issues) which you will run into with any CAD (some will be more reliable others less).
To work around this issues TechDraw allows now to define parametric vertices in 3d which are directly identifiable in 2d and therefore guarantee to avoid the topo-naming issue.
The other problem is the parametric positioning of annotations. This can be solved by regression models (eg: scikit-learn) which acts as a mapping between parameters and the relative position of measurements, annotations... .

If you are interested in this topic, it's best to get active in the freecad-forum and help with testing the latest 0.19 version (especially techdraw).