tpaviot/pythonocc-core #757: I wangt to develop a software which can select standard part like as Misumi

qunat, 2020-01-13:
which function can show 3D part in QT windows all the time

tpaviot, 2020-01-15:
what is misumi ? please be more precise with your question, there's no way to answer such a general and short question.

qunat, 2020-01-17:
thanks for your reply.I am so sorry i do not describe the question in detail MiSumi is a Japanese develop software. the software could help machine design engineer to draw a part quickly.  in fact designer only click the mouse to choose the standard part they want  then the software will draw it automatically and output step or IGES data. now i want copy it . my trouble is how to build a window  the window could show the part all the time 

if you can't get my ideal you can see these pictures in the email.  the little window which show 3D part is I want to build.

 At last thanks for your work in pythonocc .my engilish is poor  please forgive me

closed 2020-02-12 by qunat