tpaviot/pythonocc-core #758: Qt5 Error Using docker on Ubuntu

voidspace, 2020-01-14:
On some machines pythonocc with docker works fine. On other machines, pythonocc-core 0.18.1, ubuntu 18.04, pyqt5, it fails with this error:
```
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt5
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
Xw_Window created.
Traceback (most recent call last):
  File "make_some_cad.py", line 48, in <module>
    show_CAD_OCC(cube, revolution)
  File "/BLUEPRINT/BLUEPRINT/cad/cadtools.py", line 179, in show_CAD_OCC
    qt_display = QtDisplayer()
  File "/BLUEPRINT/BLUEPRINT/cad/display.py", line 41, in __init__
    self.qt_display, self.start_qt_display, menu, f_menu = init_display()
  File "/usr/lib/python3/dist-packages/OCC/Display/SimpleGui.py", line 188, in init_display
    win.canva.InitDriver()
  File "/usr/lib/python3/dist-packages/OCC/Display/qtDisplay.py", line 136, in InitDriver
    self._display.Create()
  File "/usr/lib/python3/dist-packages/OCC/Display/OCCViewer.py", line 181, in Create
    self.Init(self._window_handle)
RuntimeError: Aspect_GraphicDeviceDefinitionError
OpenGl_Window::CreateWindow: glXCreateContext failed.
```
Any suggestions as to how I can diagnose?

voidspace, 2020-01-14:
With `LIBGL_DEBUG` set. Note that on the host machine `$DISPLAY=:1`
```
root@dada05586c16:/BLUEPRINT/examples/cad# LIBGL_DEBUG=verbose python3 make_some_cad.py 
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt5
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
libGL: screen 0 does not appear to be DRI2 capable
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /root/.drirc: No such file or directory.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
Xw_Window created.
libGL: screen 0 does not appear to be DRI2 capable
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /root/.drirc: No such file or directory.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Traceback (most recent call last):
  File "make_some_cad.py", line 48, in <module>
    show_CAD_OCC(cube, revolution)
  File "/BLUEPRINT/BLUEPRINT/cad/cadtools.py", line 179, in show_CAD_OCC
    qt_display = QtDisplayer()
  File "/BLUEPRINT/BLUEPRINT/cad/display.py", line 41, in __init__
    self.qt_display, self.start_qt_display, menu, f_menu = init_display()
  File "/usr/lib/python3/dist-packages/OCC/Display/SimpleGui.py", line 188, in init_display
    win.canva.InitDriver()
  File "/usr/lib/python3/dist-packages/OCC/Display/qtDisplay.py", line 136, in InitDriver
    self._display.Create()
  File "/usr/lib/python3/dist-packages/OCC/Display/OCCViewer.py", line 181, in Create
    self.Init(self._window_handle)
RuntimeError: Aspect_GraphicDeviceDefinitionError
OpenGl_Window::CreateWindow: glXCreateContext failed.
```

tpaviot, 2020-01-15:
I guess the machines are different, it seems there's an issue with the opengl driver

voidspace, 2020-01-15:
We gave up on docker and ended up compiling from source into a venv.
For the sake of google this is how:
```
set -e

# Ensure we are working in a virtual environment
if [ ! -d .env ]; then
  python3 -m venv .env
fi
. .env/bin/activate

cd ..

if [ ! -d smesh ]; then
  git clone git://github.com/tpaviot/smesh.git
  cd smesh
  git checkout 6.7.4
  mkdir cmake-build
  cd cmake-build
  cmake ..
  make -j 4
  make install
fi

cd ..
if [ ! -d "pythonocc-core" ]; then
  git clone git://github.com/tpaviot/pythonocc-core.git
fi

cd pythonocc-core
git checkout 0.18.1
mkdir -p cmake-build
cd cmake-build
cmake .. -DPYTHONOCC_WRAP_SMESH=ON
make -j 4
make install
#cd ../test
#python3 run_tests.py
```

closed 2020-01-15 by voidspace