tpaviot/pythonocc-core #759: AIS_InteractiveObejct is missing SetSelectionPriority

Brently-Mills, 2020-01-16:
Hello,
I have been porting our pythonocc 0.18.1 project to the latest version of pythonocc. It appears that the API SetSelectionPriority is missing from the AIS_InteractiveObejct. Has this API been moved?

Thank you for providing pythonocc!

Thank you,
Brent

tpaviot, 2020-01-16:
The ```AIS_InteractiveObject::SetSelectionPriority``` method as disappeared from upstream OCCT between 6.9.0 release and 7.4.0, but I don't know exactly when, and I don't know the replacement either.

Brently-Mills, 2020-01-16:
Thank you very much for looking into this. I'll see if I can find another way to set the selection priority on this end and report back if I find anything.