tpaviot/pythonocc-core #760: How can I remove the X,Y,Z axes when export the rendered image?

siyuan2018, 2020-01-16:
Thanks for your work first! I am using OffscreenRenderer(), which is inherited from Display3d() I think, to export a front view image of a STEP file. But when I opened the window, the coordinate axes, X, Y, and Z appear automatically. I installed pythonocc by conda, so I have no experience in compiling the source code and install it from C++. I need a pure front view image, is there any way I can get rid of the axes just modifying some python code? 
Thanks so much again!

tpaviot, 2020-01-16:
By X, Y and Z axis, you mean the trihedron in the lower corner, right ? To hide the trihedron:

```display.hide_triedron()```

siyuan2018, 2020-01-17:
Thanks so much! It solved my problem to hide the triedron.
So my final target is to generate line drawing from different views, now I have the result for the front line drawing of a model (PNG file), it is:
[image]

However, I hope I can achieve such result, with red hidden line (I use FreeCAD to generate the SVG file and this is the SVG file snapshot. I understand there is a quality loss to convert it to png):
[image]

And the model seen from isometric view is as follows, and it is symmetric:
[image]

Currently my script is like:
step_reader = STEPControl_Reader()
step_reader.ReadFile('/my/path/1.step')
step_reader.TransferRoot()
myshape = step_reader.Shape()
Viewer3d = OffscreenRenderer()
Viewer3d.SetOrthographicProjection()
Viewer3d.SetModeWireFrame()
Viewer3d.View_Front()
Viewer3d.hide_triedron()
Viewer3d.SetRenderingParams()
color = Quantity_Color(0, 0, 0, 0)
r = Viewer3d.DisplayShape(myshape, color = color)

I think to achieve my final goal, I need to tune the line's property, including hidden line's property. Do you have any suggestions that how I can achieve it? Thanks so much again! I really appreciate it!

tpaviot, 2020-02-05:
See https://github.com/tpaviot/pythonocc-demos/issues/21 I think it solves the issue.