tpaviot/pythonocc-core #761: V3d_Viewer.AddZLayer API has changed and requires Graphic3d_ZLayerId as input

Brently-Mills, 2020-01-17:
Hello,
I apologize for adding one more issue. In my effort to convert from pythonocc 0.18.1 to 7.4.0. The behavior of the V3d_Viewer.AddZLayer has changed. The API now expects the type Graphic3d_ZLayerId as input although the input type should be an integer. Is there a work around for this API so I can set the layers for displaying wire frame over solid geometry .

https://www.opencascade.com/content/visibility-edges-vertexes-through-shells

Thank you,
Brent

tpaviot, 2020-01-17:
```aViewer.AddZLayer(aLayerId))```
where aLayerId is chosen among
```Graphic3d_ZLayerId_UNKNOWN # identifier for invalid ZLayer
Graphic3d_ZLayerId_Default  # default Z-layer for main presentations
Graphic3d_ZLayerId_Top  # overlay for 3D presentations which inherits Depth from previous ZLayer
Graphic3d_ZLayerId_Topmost # overlay for 3D presentations with independent Depth
Graphic3d_ZLayerId_TopOSD # overlay for 2D presentations (On-Screen-Display)
Graphic3d_ZLayerId_BotOSD
```

tpaviot, 2020-01-17:
See also https://documentation.help/Open-Cascade/occt_dev_guides__upgrade.html and especially the section **On-screen objects and ColorScale**

tpaviot, 2020-01-17:
At last, have a look at the ZLayer test code, from occt (c++) http://git.dev.opencascade.org/gitweb/?p=occt.git;a=blob;f=src/ViewerTest/ViewerTest_ViewerCommands.cxx;h=c1e1e059d5c723538aa15ae4c1ba280623b8910c;hb=HEAD#l5445

Brently-Mills, 2020-01-21:
Thank you very much for the rapid response. I did try the following but still get a type error:
ex (pseudo code):

from OCC.Core.Graphic3d import Graphic3d_ZLayerId_Topmost

V3d_Viewer..AddZLayer(Graphic3d_ZLayerId_Topmost)

TypeError: in method 'V3d_Viewer_AddZLayer', argument 2 of type 'Graphic3d_ZLayerId &'

I assume I ma missing something obvious but I do not see a way to cast to the Graphic3d_ZLayerId  type.

Thank you,
Brent

tpaviot, 2020-01-21:
Good catch, there's indeed an issue in the way ```Graphic3d_ZLayerId &``` is wrapped. A fix is being tested. Can you please provide a full example so that I can quickly test ?

Brently-Mills, 2020-01-21:
Setting the ZLayer is in the middle of our UI code -- which is complex. 
I have lifted the ZLayer call and have made a version of the SimpleGui that shows a sphere and a line. Adding the ZLayer should have the line show through the sphere. The AddZLayer API call is at line 501 in the attached version of the OCCViewer. For what it is worth this is only for wxPython.

Thank you,
Brent

Brently-Mills, 2020-01-21:
It seems I cannot upload files from inside my firewall. I'll upload it later today.

tpaviot, 2020-01-22:
The ```AddZLayer``` method has changed. It requires a  ```Graphic3d_ZLayerSettings``` as input, and outputs the ```Graphic3d_ZLayerId``` id. I committed a fix here 3348717af39b1d15ca65f3cc4399339e6bbeeade

The way you should use AddZLayer is now:
```
settings = Graphic3d_ZLayerSettings()
# then update any setting
is_ok, layer_id = my_viewer.AddZLayer(settings)
```
```is_ok``` is True if the ZLayer was successfully added, and you get the layer_id as an output.

tpaviot, 2020-01-22:
I uploaded an example here https://github.com/tpaviot/pythonocc-demos/commit/2680be2e15c8722bffd6c55d838087f75ba33ef3

The sphere is always displayed in foreground, and the box background.

Brently-Mills, 2020-01-22:
Wow. Thank you so very much!

tpaviot, 2020-01-23:
Fixed by PR #765

closed 2020-01-23 by tpaviot