tpaviot/pythonocc-core #762: All 4 vertices of an element in global coordinate system

khyatiip, 2020-01-17:
Hello,

I'm trying to find all 4 vertices of an element in the global coordinate system.
```
import ifcopenshell
import ifcopenshell.geom
from OCC.Core import BRep
from OCC.Core import BRepTools
from OCC.Core import TopAbs
from OCC.Core import TopoDS
from OCC.Core import TopExp
settings = ifcopenshell.geom.settings()
settings.set(settings.USE_PYTHON_OPENCASCADE, True)

ifc_file = ifcopenshell.open('untitled1.ifc')
    
spaces = ifc_file.by_type('IfcWindow')[0]

product = ifcopenshell.geom.create_shape(settings, spaces)
shape = TopoDS.TopoDS_Iterator(product.geometry).Value()
trsf = shape.Location().Transformation()
print (trsf.TranslationPart().Coord())
```

This only gives me one coordinate. How do I get all the vertices/coordinates? Can someone help me here with an example?

tpaviot, 2020-01-17:
The TopoDS_Iterator is IfcOpenshell specific, but I guess it can be iterated (see https://github.com/IfcOpenShell/IfcOpenShell/blob/1426c7066bee83f65a5250811ae35ac8923f0661/src/ifcopenshell-python/ifcopenshell/geom/occ_utils.py#L88). You should try something like:
```
it = TopoDS.TopoDS_Iterator(shape)
while it.More():
   shp = it.Value()
   trsf = shape.Location().Transformation()
   print (trsf.TranslationPart().Coord())
   it.Next()
```

for shp in TopoDS.TopoDS_Iterato(produ

khyatiip, 2020-01-17:
I tried this. This still gives me only one coordinate.

bricway, 2020-01-23:
Maybe this can help you ( to get the Sub-Shapes of Vertex use the Explorer with TopAbs_VERTEX):

 ```
    it = TopExp_Explorer(shape, TopAbs_VERTEX)
    while it.More():
        shp = it.Value()
        shp: TopoDS_Vertex # we get only Sub-Shapes of Vertex by the Iterator
        shp_g = BRep_Tool.Pnt(shp) # get the geometry of the Vertex ->point
        shp_g: gp_Pnt
        trsf = shape.Location().Transformation()
        print(shp,trsf.TranslationPart().Coord())
        print("Vertex-Coordinates/Global (%6.3f,%6.3f,%6.3f)" % (shp_g.X(),shp_g.Y(),shp_g.Z() ))
        print("-"*20)
        it.Next()

```