tpaviot/pythonocc-core #764: TopTools_MapIteratorOfMapOfShape could not be find in the python-interface

bricway, 2020-01-21:
Hello,

I hope my question is not so stupid, but I coudn't find the python-Interface/wrapper Class for  **TopTools_MapIteratorOfMapOfShape**.

How can I use this class  with python-occ? (Or do we have no wrapper?)

Are there any other techniques without using this Iterator-Class to iterate over a MapOfShapes?

tpaviot, 2020-01-22:
Which method/class do you use that require a ```TopTools_MapIteratorOfMapOfShape``` ?

bricway, 2020-01-22:
Thanks for your fast feedback!

Based on your query - I think - I got the point: The Class `TopTools_MapIteratorOfMapOfShape` is too elementary, that a Python-Wrapper is defined. Indeed I couldn't find a class/method in the core that requires this class as an input (Of course in the core it is used)! **Hence there is no need under this context (Input for core methods)**!

Actually I use pythonocc as a productive environment to analyze and test new algorithms (instead using c++). Hence I was looking for a python-wrapper class to iterate over the items of a Map (e.g. `MapOfShapes`). As the elementary container classes are very useful I really thought, that there must be somewhere `MapIteratorXXX`-wrappers ... especially when I read in the [Overview of the pythonOCC SWIG files generation process ](https://github.com/tpaviot/pythonocc-generator/blob/master/Documentation.md) "_Class names may seem ugly, for instance 'BRepPrimAPI_MakeBox', '**TopTools_MapIteratorOfMapOfShape**' etc. ..._"
... so I thought I make something wrong regarding the `import` statements.