tpaviot/pythonocc-core #769: How to extract mesh vertices global coordinates

HerveGlz, 2020-01-28:
Hi all,
From the python script **examples/core_simple_mesh.py** (, I am trying to extract mesh vertices global coordinates so that I can inject them inside the oriented bounding box script (https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_geometry_oriented_bounding_box.py) to replace the random vertices creation step of the latter.

Is there someone who could give me a hand on it ? It will help me a lot.

Thanks in advance,
Hervé

HerveGlz, 2020-01-29:
To illustrate a bit more my question and the context, here is the code I succeeded to write:
```Python
from OCC.Core.BRep import BRep_Builder, BRep_Tool
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox

from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeVertex

from OCC.Core.BRepMesh import BRepMesh_IncrementalMesh
from OCC.Core.TopExp import TopExp_Explorer
from OCC.Core.TopoDS import TopoDS_Compound, topods_Face
from OCC.Core.TopAbs import TopAbs_FACE
from OCC.Core.TopLoc import TopLoc_Location
from OCC.Core.gp import gp_Pnt, gp_Ax2, gp_Dir, gp_XYZ

from OCC.Display.SimpleGui import init_display
from OCC.Extend.DataExchange import read_step_file

# For Bounding Box line and transparency modification
from OCC.Core.AIS import AIS_Shape

from OCC.Core.BRepBndLib import brepbndlib_AddOBB
from OCC.Core.Bnd import Bnd_OBB
from OCC.Core.GProp import GProp_GProps
from OCC.Core.BRepGProp import brepgprop_VolumeProperties

from OCC.Extend.TopologyUtils import TopologyExplorer

display, start_display, add_menu, add_function_to_menu = init_display()
display.SetSelectionModeVertex()

#======================================================================================
def MinimumBBox(filepath):
    shape = read_step_file(filepath)
    display.DisplayShape(shape)

    obb = Bnd_OBB()

    # We call the mesh function to obtain the vertices point cloud
    VerticesContainer = simple_mesh(shape, extractionMode="vertices")

    #---------------------------------------------------------------------------------
    nbMeshedSurface = len(VerticesContainer)
    # ITERATION OVER ALL FACES OF THE MESH
    for surface in range(nbMeshedSurface):
        vertices = VerticesContainer[surface]
        nbVerticesFace = len(vertices)

        # FOR EACH FACES WE COLLECT VERTICES
        for i in range(nbVerticesFace):
            print(vertices[i])
            p = BRepBuilderAPI_MakeVertex(vertices[i]).Shape()
            brepbndlib_AddOBB(p, obb)  # Vertex transfer to the oriented bounding box
    # ---------------------------------------------------------------------------------

    #------------------------------------------------------------------------------------------
    def _ConvertBndToShape(theBox):

      #-----------------------------------------------------------------------------
      aBaryCenter = theBox.Center()
      aXDir = theBox.XDirection()
      aYDir = theBox.YDirection()
      aZDir = theBox.ZDirection()
      aHalfX = theBox.XHSize()
      aHalfY = theBox.YHSize()
      aHalfZ = theBox.ZHSize()

      ax = gp_XYZ(aXDir.X(), aXDir.Y(), aXDir.Z())
      ay = gp_XYZ(aYDir.X(), aYDir.Y(), aYDir.Z())
      az = gp_XYZ(aZDir.X(), aZDir.Y(), aZDir.Z())
      p = gp_Pnt(aBaryCenter.X(), aBaryCenter.Y(), aBaryCenter.Z())
      anAxes = gp_Ax2(p, gp_Dir(aZDir), gp_Dir(aXDir))
      anAxes.SetLocation(gp_Pnt(p.XYZ() - ax*aHalfX - ay*aHalfY - az*aHalfZ))
      aBox = BRepPrimAPI_MakeBox(anAxes, 2.0*aHalfX, 2.0*aHalfY, 2.0*aHalfZ).Shape()
      #-----------------------------------------------------------------------------

      # VOLUME COMPUTATION
      props = GProp_GProps()
      brepgprop_VolumeProperties(aBox, props)
      # Get inertia properties
      volume = props.Mass()
      print(f"The Minimal Bounding Box Volume is: {volume}mm3")

      # Bounding box appearance modification
      ais_shp = AIS_Shape(aBox)
      ais_shp.SetWidth(4)
      ais_shp.SetTransparency(0.60)

      # Bounding box display
      ais_context = display.GetContext()
      ais_context.Display(ais_shp, True)

      return ais_shp

    # We Call the Oriented Bounding Box Function
    _ConvertBndToShape(obb)

    display.FitAll()
    start_display()

#======================================================================================
def simple_mesh(shape, extractionMode):
    VerticesContainer = []

    model = TopologyExplorer(shape)

    #---------------
    # Mesh the shape
    #---------------
    deflection = 0.6
    BRepMesh_IncrementalMesh(shape, deflection)
    builder = BRep_Builder()
    comp = TopoDS_Compound()
    builder.MakeCompound(comp)

    bt = BRep_Tool()
    ex = TopExp_Explorer(shape, TopAbs_FACE)

    if extractionMode == "vertices":

        while ex.More():
            face = topods_Face(ex.Current())
            location = TopLoc_Location()
            facing = bt.Triangulation(face, location)
            vertices = facing.Nodes()
            #tri = facing.Triangles()

            # We populate the list VerticesContainer
            VerticesContainer.append(vertices)

            ex.Next()
    return VerticesContainer

if __name__ == '__main__':
    MinimumBBox("C:/Users/Herve/Desktop/TEST_3D/S_Bois.stp")

# cube4.STEP
# Di Pietro Motor V2_STEP.stp
# Fixation de Ski.stp
# Stabilisateur Reflex -3Kg.stp
# Ensemble Circuit Puissance.stp
```Python

HerveGlz, 2020-01-29:
[image]
[image]

My actual problem is the bounding box is not optimal/minimal one since it could be smaller...

HerveGlz, 2020-01-31:
Hi all,
I've recently found that there is an option (theIsOptimal: bool) to specify if the OBB needs to be optimal or not .
But I don't know how to access it and specify it for my "**brepbndlib_AddOBB(p, obb)**"

[image]

Am I on the right way ?

HerveGlz, 2020-01-31:
I finally figured out how it works! If some people are interested in discovering how to solve this problem, here is the solution to get a good OBB which works (until now):

```Python
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox

from OCC.Core.gp import gp_Pnt, gp_Ax2, gp_Dir, gp_XYZ

from OCC.Display.SimpleGui import init_display
from OCC.Extend.DataExchange import read_step_file

# For Bounding Box line and transparency modification
from OCC.Core.AIS import AIS_Shape

from OCC.Core.BRepBndLib import brepbndlib_AddOBB
from OCC.Core.Bnd import Bnd_OBB
from OCC.Core.GProp import GProp_GProps
from OCC.Core.BRepGProp import brepgprop_VolumeProperties

display, start_display, add_menu, add_function_to_menu = init_display()
display.SetSelectionModeVertex()

#====================================================================================================================
def MinimumBBox(filepath):
    shape = read_step_file(filepath)
    display.DisplayShape(shape)

    OBB = Bnd_OBB()
    IsTriangulationUsed = True
    IsOptimal = True
    IsShapeToleranceUsed = False

    brepbndlib_AddOBB(shape, OBB, IsTriangulationUsed, IsOptimal, IsShapeToleranceUsed)  # Vertex transfer to the oriented bounding box

    #------------------------------------------------------------------------------------------
    def _ConvertBndToShape(theBox):

      #-----------------------------------------------------------------------------
      aBaryCenter = theBox.Center()
      aXDir = theBox.XDirection()
      aYDir = theBox.YDirection()
      aZDir = theBox.ZDirection()
      aHalfX = theBox.XHSize()
      aHalfY = theBox.YHSize()
      aHalfZ = theBox.ZHSize()

      ax = gp_XYZ(aXDir.X(), aXDir.Y(), aXDir.Z())
      ay = gp_XYZ(aYDir.X(), aYDir.Y(), aYDir.Z())
      az = gp_XYZ(aZDir.X(), aZDir.Y(), aZDir.Z())
      p = gp_Pnt(aBaryCenter.X(), aBaryCenter.Y(), aBaryCenter.Z())
      anAxes = gp_Ax2(p, gp_Dir(aZDir), gp_Dir(aXDir))
      anAxes.SetLocation(gp_Pnt(p.XYZ() - ax*aHalfX - ay*aHalfY - az*aHalfZ))
      aBox = BRepPrimAPI_MakeBox(anAxes, 2.0*aHalfX, 2.0*aHalfY, 2.0*aHalfZ).Shape()
      #-----------------------------------------------------------------------------

      # VOLUME COMPUTATION
      props = GProp_GProps()
      brepgprop_VolumeProperties(aBox, props)
      # Get inertia properties
      volume = props.Mass()
      print(f"The Minimal Bounding Box Volume is: {volume}mm3")

      # Bounding box appearance modification
      ais_shp = AIS_Shape(aBox)
      ais_shp.SetWidth(4)
      ais_shp.SetTransparency(0.60)

      # Bounding box display
      ais_context = display.GetContext()
      ais_context.Display(ais_shp, True)

      return ais_shp

    # We Call the Oriented Bounding Box Function
    _ConvertBndToShape(OBB)

    display.FitAll()
    start_display()

if __name__ == '__main__':
    MinimumBBox("C:/Users/Herve/Desktop/TEST_3D/cube4.STEP")

# cube4.STEP
# Di Pietro Motor V2_STEP.stp
# Fixation de Ski.stp
# Stabilisateur Reflex -3Kg.stp
# Ensemble Circuit Puissance.stp
```

tpaviot, 2020-02-03:
Gret, thank you. I updated the related example at pythonocc-demos see https://github.com/tpaviot/pythonocc-demos/commit/1f82208f6282364407ee7aa95467ed78a3de5a70

closed 2020-02-03 by tpaviot