tpaviot/pythonocc-core #772: ImportError: DLL load failed: The specified module could not be found

khyatiip, 2020-02-06:
Hello,

When i run my python script using Spyder, no error is raised. It works perfectly fine. But when I run the same python script using a batch script, I get the following error. How do I solve this?
```
C:\Users\KHYATI\Desktop\Thesis>python "C:\Users\KHYATI\Desktop\Thesis\_FenestrationSurface.py"
Traceback (most recent call last):
  File "C:\Users\KHYATI\Desktop\Thesis\_FenestrationSurface.py", line 3, in <module>
    from OCC.Core import TopoDS
  File "C:\Users\KHYATI\AppData\Local\Programs\Python\Python37\lib\site-packages\OCC\Core\TopoDS.py", line 18, in <module>
    from . import _TopoDS
ImportError: DLL load failed: The specified module could not be found.
```

qunat, 2020-02-14:
can you show your code?

tpaviot, 2020-02-15:
I guess there's something wrong with your PATH or PYTHONPATH

tpaviot, 2020-02-28:
solved ?

lanix516, 2020-04-23:
Hello,
    have you solved this?

tpaviot, 2020-04-23:
no

lanix516, 2020-04-28:
change python 3.8
conda install -c tpaviot  pythonocc-core=7.4.0
It works very well。