tpaviot/pythonocc-core #774: How to delete display shape

qunat, 2020-02-12:
This is my code :
    def show_3d_model(self,filename=""):#显示3D数据 
        "获取选择零件名称  获取路径" Get part path
        #获取零件名称
        pass
        self.partname=""
        #获取相应零件的路径
        self.partpath=os.getcwd()
        self.partpath=self.partpath+"/3Ddata"+"/"+self.partname+".step"
        self.canva._display.EraseAll()# Delete part
        #self.canva._display.display_triedron()
        #self.canva._display.set_bg_gradient_color(background_gradient_color1, background_gradient_color2)
        try:
            shape=read_step_file(self.partpath)
            #shape = BRepPrimAPI_MakeBox(10, 10, 10).Shape()
            self.canva._display.DisplayShape(shape)
            self.canva._display.FitAll()
        except:
            pass
        return filename
    I used  '.EraseAll()'  to delete display shape but the coordinate system is deleted too.

tpaviot, 2020-02-28:
What do you mean with "coordinate system erased", the trihedron ? If it disappears, just display it with ```display.display_triedron()```

qunat, 2020-03-06:
Thank  you I get it

closed 2020-03-06 by qunat