tpaviot/pythonocc-core #775: incorrectly imported step files

michele-arrival, 2020-02-13:
Hi, 

There seems to be a discrepancy when loading some step files, in that not all shapes are fetched by the function "read_step_file" (in fact only the first one is), where instead "read_step_file_with_names_colors" loads all shapes correctly.

The file that reproduces the error can be obtained from this dataset: https://deep-geometry.github.io/abc-dataset/
Any of the step files in the dataset triggers the issue, however to cut it short one can simply download the ste file in the front page: https://deep-geometry.github.io/abc-dataset/data/00000050_80d90bfdd2e74e709956122a_step_000.step 

To reproduce one can run "core_load_step_ap203.py" from the demos repo with this file. The visualized part will be like this - both when doing "import_as_one_shape" and "import_as_multiple_shapes"
[image]

whereas both samples `core_load_step_ap203_ocaf.py` and `core_load_step_with_colors.py` load the right number of shapes:
[image]
[image]

cheers

qunat, 2020-02-14:
som step file you want to load must be saved stp not step

tpaviot, 2020-02-15:
Thanks for the report , investigating

closed 2020-02-15 by tpaviot

jf---, 2020-02-16:
> The file that reproduces the error can be obtained from this dataset: https://deep-geometry.github.io/abc-dataset/

@michele-arrival that is an **unbelievable** resource, *wow*! 
can we perhaps invite you to write a post on [pythonocc.org](http://www.pythonocc.org/) on how you are using this dataset and what the connection to `pythonocc` is? 
I'm incredibly curious ;)

michele-arrival, 2020-02-28:
> @michele-arrival that is an **unbelievable** resource, _wow_!
> can we perhaps invite you to write a post on [pythonocc.org](http://www.pythonocc.org/) on how you are using this dataset and what the [...]

Hi and thanks for the interest! 
I haven't entirely defined what the use of pythonocc will be for me. For the time being i was looking for a python module to render parametric models and pythonocc seemed to be a good choice for that. In the future I will possibly look at geometric - or even semantic - classification of the parts and identify what they are useful for. It's an ongoing project, but I'm happy to take you up on the offer in the future once I have a clearer picture. Thanks indeed for the proposal!

@tpaviot Thanks for the fix in the new release!

commit 078bd3cf (2020-02-15): Fixed read_step_file if more than one shape at root, fix #775