tpaviot/pythonocc-core #778: Segmentation fault: Sewing

e3dEF, 2020-02-21:
Hi,
I get a segmentation fault, when calling FreeEdge of a BRepBuilderAPI_Sewing() object using the current beta version (conda install, pythonocc-core 7.4.0beta2 py36h6bb024c_0). I tested also in 0.18.2 where it works. 
Maybe you are aware of that fault already.

Here is the Dump of the sew:
```
 Number of input shapes      : 30
 Number of actual shapes     : 30
 Number of Bounds            : 86
 Number of Sections          : 86
 Number of Edges             : 65
 Number of Vertices          : 40
 Number of Nodes             : 40
 Number of Free Edges        : 56
 Number of Contigous Edges   : 1
 Number of Multiple Edges    : 8
 Number of Degenerated Edges : 0
```
 
Here the result of 0.18.2
```
class<'TopoDS_Edge'; id:2042279761>
class<'TopoDS_Edge'; id:2042253057>
class<'TopoDS_Edge'; id:2042258465>
...
```

tpaviot, 2020-02-24:
Can you please give any clue to reproduce the issue ?

e3dEF, 2020-02-24:
I think I found MY mistake. I tried to access the edges in range(sew.NbFreeEdges()). Instead, I have to loop in range(1, sew.NbFreeEdges() + 1). The difference between 0.18.2 and 7.4 was that 0.18.2 gave a RuntimeError: Standard_OutOfRange that was catched by a try statement. 7.4. raised the segmentation fault.

rainman110, 2020-02-24:
@tpaviot A segfault should never occur. Instead pythonocc should catch the c++ exception and convert it to a runtime error.

rainman110, 2020-02-24:
@e3dEF What OS are you using? I noticed some problems wrt exception handling on macos.

tpaviot, 2020-02-24:
@rainman110 pythonocc only catches occt exceptions. But if occt fails without raising any exception (i.e. occt segfaults), then pythonocc will segfault as well. That is to say that, if pythonocc segfaults, then it comes from an uncatched occt exception and should be reported upstream. As far as I know, catching occt exception works quite well: all calls to occt methods are put within a try/catch block (see https://github.com/tpaviot/pythonocc-core/blob/master/src/SWIG_files/common/ExceptionCatcher.i)

tpaviot, 2020-02-24:
@e3dEF just to check, can you please post the code you've been testing with ?

rainman110, 2020-02-24:
@tpaviot the pythonocc exception catcher is correct. The problems I noticed is a macos / compiler specific thing than also happens on other pure c++ code. 

From what I have seen, it seems to happen when using the conda-build compiler.

adam-urbanczyk, 2020-02-24:
@rainman110 maybe it's just noise but I have seen similar issues when doing custom builds on older PythonOCC versions too https://github.com/CadQuery/cadquery/issues/207 . I know too little about MacOS to figure that one out.

e3dEF, 2020-02-24:
My OS is Ubuntu 18.04.
Here is the code
```
from OCC.Core.BRepTools import breptools_Read
from OCC.Core.TopoDS import TopoDS_Shape
from OCC.Core.BRep import BRep_Builder
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_Sewing

sew = BRepBuilderAPI_Sewing()

for i in range(2):
    shape = TopoDS_Shape()
    builder = BRep_Builder()
    breptools_Read(shape, str(i) + ".brep", builder)
    sew.Add(shape)

sew.Perform()
sew.Dump()

print(sew.FreeEdge(0))
```
The opened breps are 3 planar faces. I could provide the files, if necessary (3kB).

tpaviot, 2020-02-25:
@adam-urbanczyk @rainman110 Exception handling is unittested, see https://github.com/tpaviot/pythonocc-core/blob/master/test/core_wrapper_features_unittest.py#L490 If necessary, I can extend the number of tests, but they actually pass on all platforms.

@e3dEF Yes, please, can you attach brep files as well

e3dEF, 2020-02-25:
Files:
[BREPs.zip](https://github.com/tpaviot/pythonocc-core/files/4248656/BREPs.zip)

tpaviot, 2020-02-25:
Here is the result on my machine:
```bash

                        Informations                        
 ===========================================================
 
 Number of input shapes      : 2
 Number of actual shapes     : 2
 Number of Bounds            : 9
 Number of Sections          : 9
 Number of Edges             : 9
 Number of Vertices          : 9
 Number of Nodes             : 9
 Number of Free Edges        : 9
 Number of Contigous Edges   : 0
 Number of Multiple Edges    : 0
 Number of Degenerated Edges : 0
 ===========================================================
 
Erreur de segmentation (core dumped)
```

I looked at the c++ code, an exception should be raised/catched (from ```BRepBuilderAPI_Sewing.cxx```):
```c++
//=======================================================================
//function : FreeEdge
//purpose  : 
//=======================================================================

const TopoDS_Edge& BRepBuilderAPI_Sewing::FreeEdge(const Standard_Integer index) const
{
  Standard_OutOfRange_Raise_if(index < 0 || index > NbFreeEdges(), "BRepBuilderAPI_Sewing::FreeEdge");
  return TopoDS::Edge(myFreeEdges(index));
}
```

The ```Standard_OutOfRange_Raise_if``` macro is defined in ```Standard_OutOfRange.hxx```, see:
```c++
#if !defined No_Exception && !defined No_Standard_OutOfRange
  #define Standard_OutOfRange_Raise_if(CONDITION, MESSAGE) Standard_OutOfRange_Always_Raise_if(CONDITION, MESSAGE)
#else
  #define Standard_OutOfRange_Raise_if(CONDITION, MESSAGE)
#endif
```
My feeling is that the occt exception is not raised, @rainman110  your opinion ?

rainman110, 2020-02-25:
Just looking at this code. Shouldn't it be
```index<1``` ?

tpaviot, 2020-02-25:
Well,  i ran another with ```gp_Dir```, where ```Standard_OutOfRange_Raise_if``` is used:

```c++
inline void gp_Dir::SetCoord (const Standard_Integer Index,
			      const Standard_Real Xi)
{
  Standard_Real X = coord.X();
  Standard_Real Y = coord.Y();
  Standard_Real Z = coord.Z();
  Standard_OutOfRange_Raise_if (Index < 1 || Index > 3, "gp_Dir::SetCoord() - index is out of range [1, 3]");
  if      (Index == 1)  X = Xi;
  else if (Index == 2)  Y = Xi;
  else                  Z = Xi;
  Standard_Real D = sqrt (X * X + Y * Y + Z * Z);
  Standard_ConstructionError_Raise_if (D <= gp::Resolution(), "gp_Dir::SetCoord() - result vector has zero norm");
  coord.SetX(X / D);
  coord.SetY(Y / D);
  coord.SetZ(Z / D);
}
```

and here is what happens with python:
```python
>>> from OCC.Core.gp import gp_Dir
>>> d = gp_Dir(1, 2, 3)
>>> d.SetCoord(0, 2.5)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Standard_OutOfRange
gp_Dir::SetCoord() - index is out of range [1, 3]
wrapper details:
  * symname: gp_Dir_SetCoord
  * wrapname: _wrap_gp_Dir_SetCoord__SWIG_0
  * fulldecl: void gp_Dir::SetCoord(Standard_Integer const,Standard_Real const)
```

This seems to work correctly here.

rainman110, 2020-02-25:
It's the issue in case index==0. This bug should be reported to the occt Mantis.

tpaviot, 2020-02-25:
Yes, it should be ```index<1``` or  ```index<=0```, but I tested with ```FreeEdge(-1)``` and ```FreeEdge(10)``` (there are 9 free edges reported by ```NbFreeEdges```), and it also segfaults

rainman110, 2020-02-25:
This is indeed very strange. It's hard to guess, what the actual problem is.

tpaviot, 2020-02-25:
Here is the smallest example that reproduces the issue
``` python
>>> from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_Sewing
>>> s = BRepBuilderAPI_Sewing()
>>> s.FreeEdge(0)
Erreur de segmentation (core dumped)
```
I'm curious to see what would be the result of the same example written in c++

tpaviot, 2020-02-25:
I tried the following example
``` c++
#include <BRepBuilderAPI_Sewing.hxx>
#include <Standard_Failure.hxx>

int main() {
	BRepBuilderAPI_Sewing s;
	try {
            s.FreeEdge(-1);
        }
       catch(Standard_Failure const& error) {
	    printf("error");
        }
	return 1;
}
```
compiled using
``` bash
g++ -I /opt/build/occt740/include/opencascade/sewing_test.cpp -L /opt/build/occt740/lib -lTKTopAlgo -lTKernel -o sewing_test
```
and result is
``` bash
$ ./sewing_test
Erreur de segmentation (core dumped)
```
that is to say, the same that python. Something prevent the exception to be raised.

Note : valgrind and gdb does not tell anything about the issue. For instance, the valgrind stack is
``` bash
$ valgrind --leak-check=full --show-leak-kinds=all ./sewing_test 
==13077== Memcheck, a memory error detector
==13077== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==13077== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==13077== Command: ./sewing_test
==13077== 
==13077== Invalid read of size 8
==13077==    at 0x5043C5A: BRepBuilderAPI_Sewing::FreeEdge(int) const (in /opt/build/occt740/lib/libTKTopAlgo.so.7.4.0)
==13077==    by 0x10B302: main (in /home/thomas/sewing_test)
==13077==  Address 0xfffffffffffffff0 is not stack'd, malloc'd or (recently) free'd
==13077== 
==13077== 
==13077== Process terminating with default action of signal 11 (SIGSEGV)
==13077==  Access not within mapped region at address 0xFFFFFFFFFFFFFFF0
==13077==    at 0x5043C5A: BRepBuilderAPI_Sewing::FreeEdge(int) const (in /opt/build/occt740/lib/libTKTopAlgo.so.7.4.0)
==13077==    by 0x10B302: main (in /home/thomas/sewing_test)
==13077==  If you believe this happened as a result of a stack
==13077==  overflow in your program's main thread (unlikely but
==13077==  possible), you can try to increase the size of the
==13077==  main thread stack using the --main-stacksize= flag.
==13077==  The main thread stack size used in this run was 8388608.
==13077== 
==13077== HEAP SUMMARY:
==13077==     in use at exit: 47,056 bytes in 93 blocks
==13077==   total heap usage: 94 allocs, 1 frees, 119,760 bytes allocated
==13077== 
==13077== 4 bytes in 1 blocks are still reachable in loss record 1 of 29
==13077==    at 0x4C31B25: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==13077==    by 0x53C1A70: Standard_MMgrRaw::Allocate(unsigned long) (in /opt/build/occt740/lib/libTKernel.so.7.4.0)
==13077==    by 0x53D63F4: TCollection_AsciiString::TCollection_AsciiString() (in /opt/build/occt740/lib/libTKernel.so.7.4.0)
==13077==    by 0x5388423: ??? (in /opt/build/occt740/lib/libTKernel.so.7.4.0)
==13077==    by 0x4010732: call_init (dl-init.c:72)
==13077==    by 0x4010732: _dl_init (dl-init.c:119)
==13077==    by 0x40010C9: ??? (in /lib/x86_64-linux-gnu/ld-2.27.so)
==13077== 
==13077== 4 bytes in 1 blocks are still reachable in loss record 2 of 29
==13077==    at 0x4C31B25: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==13077==    by 0x53C1A70: Standard_MMgrRaw::Allocate(unsigned long) (in /opt/build/occt740/lib/libTKernel.so.7.4.0)
==13077==    by 0x53D63F4: TCollection_AsciiString::TCollection_AsciiString() (in /opt/build/occt740/lib/libTKernel.so.7.4.0)
==13077==    by 0x538844C: ??? (in /opt/build/occt740/lib/libTKernel.so.7.4.0)
==13077==    by 0x4010732: call_init (dl-init.c:72)
[... 224 lines cut ...]
==13077==    by 0x4010732: _dl_init (dl-init.c:119)
==13077==    by 0x40010C9: ??? (in /lib/x86_64-linux-gnu/ld-2.27.so)
==13077== 
==13077== 42,640 bytes in 41 blocks are still reachable in loss record 29 of 29
==13077==    at 0x4C31B25: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==13077==    by 0x5CD22FE: __new_exitfn (cxa_atexit.c:109)
==13077==    by 0x5CD2488: __internal_atexit (cxa_atexit.c:40)
==13077==    by 0x5CD2488: __cxa_atexit (cxa_atexit.c:66)
==13077==    by 0x4010732: call_init (dl-init.c:72)
==13077==    by 0x4010732: _dl_init (dl-init.c:119)
==13077==    by 0x40010C9: ??? (in /lib/x86_64-linux-gnu/ld-2.27.so)
==13077== 
==13077== LEAK SUMMARY:
==13077==    definitely lost: 0 bytes in 0 blocks
==13077==    indirectly lost: 0 bytes in 0 blocks
==13077==      possibly lost: 0 bytes in 0 blocks
==13077==    still reachable: 47,056 bytes in 93 blocks
==13077==                       of which reachable via heuristic:
==13077==                         multipleinheritance: 384 bytes in 6 blocks
==13077==         suppressed: 0 bytes in 0 blocks
==13077== 
==13077== For counts of detected and suppressed errors, rerun with: -v
==13077== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Erreur de segmentation (core dumped)
```

rainman110, 2020-02-25:
@tpaviot Did you build occt on your own or do you use the conda packages?

tpaviot, 2020-02-25:
It's mine, compile in ```RelWithDebInfo``` mode

rainman110, 2020-02-25:
This is very confusing. Especially, as the exception is thrown and catched the case of gp_Dir.

There is one important difference though. In the gp_Dir case, the function is inline. Therefore the code will not be thrown over library boundaries. Also so it uses the same compiler as the executable..

But this is just guesswork.

rainman110, 2020-02-25:
@tpaviot Strange, I can't reproduce it. I just build occt in debug mode on ubuntu 18.04 and used your c++ test code and the exception is catched.

I'll try now using release mode.

rainman110, 2020-02-25:
If occt is compiled in *RELEASE* mode, I indeed get a segfault!

tpaviot, 2020-02-25:
@rainman110 ah, interesting feedback. Do you use the default ubuntu 18.04 compiler ?

rainman110, 2020-02-25:
yes, gcc 7.something

tpaviot, 2020-02-25:
On Windows, the following script
```python
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_Sewing
>>> s = BRepBuilderAPI_Sewing()
>>> s.FreeEdge(0)
```
also fails. No exception is catched. Using the conda binary package.

rainman110, 2020-02-25:
More debugging:

it seems, that ```No_Exception``` is defined in release mode. Have a look into Standard_OutOfRange.hxx, where this is used.

So, exceptions seem to be disabled at some point.

rainman110, 2020-02-25:
In CMakeLists.txt:

```cmake
# option disabling OCCT exceptions in Release builds (No_Exception)
if (NOT DEFINED BUILD_RELEASE_DISABLE_EXCEPTIONS)
  set (BUILD_RELEASE_DISABLE_EXCEPTIONS ON CACHE BOOL "${BUILD_RELEASE_DISABLE_EXCEPTIONS_DESCR}")
endif()
```

This explains the behaviour.

rainman110, 2020-02-25:
I am rebuilding now opencascade with exceptions enabled. I hope, this bug is resolved then.

rainman110, 2020-02-25:
https://github.com/DLR-SC/tigl-conda/commit/98f109db438e8a023950bb49adccb87d26ea12f1

rainman110, 2020-02-25:
@tpaviot , @e3dEF I updated the opencascade 7.4.0 packages at dlr-sc. In theory, this issue should now be fixed (partially, in case index=0 there is still a occt bug, FreeEdge(0) is therefore still crashing, but not FreeEdge(-1) anymore). 

You just need to update the opencascade package.

e3dEF, 2020-02-26:
Works. Thank you.

closed 2020-02-26 by e3dEF

tpaviot, 2020-02-26:
@rainman110 Successfully tested, I added a test case to the testing suite.

Be careful @looooo , IMO the occt740 package on conda-forge should be modified the same way, because the unittest I just added makes it segfault.

looooo, 2020-02-26:
https://github.com/conda-forge/occt-feedstock/pull/35