tpaviot/pythonocc-core #780: Sewing: creation of common topology fails (v7.4)

e3dEF, 2020-02-26:
When sewing two separate faces, that share an edge geometrically, I won't get a common topology. As shown in the Dump after sewing, the vertices are not merged, despite having the same coordinates. I tested nearly every possible combination of "set" parameters of the sewing class, but there is no creation of common vertices or a common edge. 
[breps.zip](https://github.com/tpaviot/pythonocc-core/files/4257104/breps.zip)

from OCC.Core.BRepTools import breptools_Read
from OCC.Core.TopoDS import TopoDS_Shape
from OCC.Core.BRep import BRep_Builder
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_Sewing
from OCCUtils.Topology import Topo, dumpTopology

f1 = TopoDS_Shape()
breptools_Read(f1, "0.brep", BRep_Builder())

f2 = TopoDS_Shape()
breptools_Read(f2, "1.brep", BRep_Builder())

sew = BRepBuilderAPI_Sewing()
sew.Add(f1)
sew.Add(f2)
sew.SetNonManifoldMode(True)
sew.Perform()

for face in Topo(sew.SewedShape()).faces():
    dumpTopology(face)
```

```
Face: 98838301
..Wire: 98471773
....Edge: 99176285
......<Vertex 96824749: 0.0 0.6 0.3>
......<Vertex 99176093: 0.5 0.6 0.3>
....Edge: 98473245
......<Vertex 99115933: 0.0 0.6 1.0>
......<Vertex 96824749: 0.0 0.6 0.3>
....Edge: 98472893
......<Vertex 99176093: 0.5 0.6 0.3>
......<Vertex 98472701: 0.8 0.6 0.3>
....Edge: 98475837
......<Vertex 98475645: 1.0 0.6 1.0>
......<Vertex 99115933: 0.0 0.6 1.0>
....Edge: 98468541
......<Vertex 98472701: 0.8 0.6 0.3>
......<Vertex 98475389: 1.0 0.6 0.3>
....Edge: 98468125
......<Vertex 98475389: 1.0 0.6 0.3>
......<Vertex 98475645: 1.0 0.6 1.0>
..Wire: 98470077
....Edge: 98471421 <----------------------------------------------------
......<Vertex 99176093: 0.5 0.6 0.3> <--------------------------------
......<Vertex 98471229: 0.5 0.6 0.5> <--------------------------------
..Wire: 98838189
....Edge: 98481709
......<Vertex 98472701: 0.8 0.6 0.3>
......<Vertex 98470253: 0.8 0.6 0.5>

Face: 96824637
..Wire: 98836765
....Edge: 96825277
......<Vertex 98523229: 0.5 1.0 0.3>
......<Vertex 98523437: 0.5 0.6 0.3>
....Edge: 98837741 <----------------------------------------------------
......<Vertex 98523437: 0.5 0.6 0.3> <--------------------------------
......<Vertex 98837533: 0.5 0.6 0.5> <--------------------------------
....Edge: 98837149
......<Vertex 98837005: 0.5 1.0 0.5>
......<Vertex 98837533: 0.5 0.6 0.5>
....Edge: 98837501
......<Vertex 98523229: 0.5 1.0 0.3>
......<Vertex 98837005: 0.5 1.0 0.5>
```

```

rainman110, 2020-02-26:
I think, this question should better be asked at the OCCT discussion forum.

e3dEF, 2020-03-18:
Answered in OCC Forum:
[https://www.opencascade.com/content/sewing-creation-common-topology-fails](url)

Sewing does not consider seam edges.

closed 2020-03-18 by e3dEF