tpaviot/pythonocc-core #781: Can't init SimpleGui

EmJay276, 2020-02-27:
Hey,

if I try to init a SimpleGui Window:
```
from OCC.Display.SimpleGui import init_display
display, start_display, add_menu, add_function_to_menu = init_display()
```

I get the following error:
```
Traceback (most recent call last):
  File "E:/xxx/xxx.py", line 15, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "C:\Anaconda3\envs\xxx\lib\site-packages\OCC\Display\SimpleGui.py", line 189, in init_display
    win.canva.qApp = app
AttributeError: can't set attribute
```

I checked line 189 in SimpleGui.py and they are already the same object before
```
In[2]: win.canva.qApp
Out[2]: <PyQt5.QtWidgets.QApplication at 0x259fb886438>
app
Out[3]: <PyQt5.QtWidgets.QApplication at 0x259fb886438>
```

If I commend out line 189 it works fine.

I'm using:
pythonocc-core 0.18.2
qt 5.9.7

Any suggestions? Thanks :-)

tpaviot, 2020-02-27:
I suggest moving up to latest 7.4.0 release.

tpaviot, 2020-02-27:
It's strange, it's the first time this issue is reported whereas the code is more than 5 years old. Do you use a conda package or did you build pythonocc by your own ?

EmJay276, 2020-02-27:
I used the conda package (cadquery package)
I'm not sure if I can upgrade to 7.4.0, cause I'm using cadquery. This isn't already moved to a newer version.

tpaviot, 2020-02-28:
These are not packages officially supported, I suggest you ask the question to the cadquery team/issue tracker.

closed 2020-02-28 by tpaviot