tpaviot/pythonocc-core #782: How to get the oriented bounding box of multiple shapes

twut, 2020-02-27:
Hi Everyone,

In ShapeFactory.py, there is a function to get the oriented_boundingbox of one single input shape.
However, I would like to calculate the oriented_boundingbox of a group of shapes.
Is there a way to do this or to merge shapes into a single shape?
Any suggestions are appreciated!

```python
def get_oriented_boundingbox(shape, optimal_OBB=True):
    """ return the oriented bounding box of the TopoDS_Shape `shape`

    Parameters
    ----------

    shape : TopoDS_Shape or a subclass such as TopoDS_Face
        the shape to compute the bounding box from
    optimal_OBB : bool, True by default. If set to True, compute the
        optimal (i.e. the smallest oriented bounding box). Optimal OBB is
        a bit longer.
    Returns
    -------
        a list with center, x, y and z sizes

        a shape
    """

```

adam-urbanczyk, 2020-02-27:
Construct `TopoDS_Compound` from your shapes, e.g. like so https://github.com/CadQuery/cadquery/blob/3d9041e98bdd9312118c4e2abfc70317ee892f01/cadquery/occ_impl/shapes.py#L1583

tpaviot, 2020-02-28:
@twut there's a function in TopologyUtils ```list_of_shapes_to_compound``` that makes it, see https://github.com/tpaviot/pythonocc-core/blob/master/src/Extend/TopologyUtils.py#L645

You just have to pass a list of shapes to this function.

closed 2020-02-28 by tpaviot

twut, 2020-02-28:
Many thanks! Works perfect for me