tpaviot/pythonocc-core #783: pythonocc-core installation issue

sohertz, 2020-03-03:
Hello, I am currently installing pythonocc but face several issues. I am following the INSTALL instructions (https://github.com/tpaviot/pythonocc-core/blob/master/INSTALL.md)

I crate a local copy of the repository, create the "cmake-build" folder and proceed to the first "cmake"

`D:mypath> cd pythonocc-core`
`D:mypath\pythonocc-core> mkdir cmake-build`
`D:mypath\pythonocc-core> cd cmake-build `
`D:mypath\pythonocc-core\cmake-build> cmake ..`

> -- Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.18362.
> --  /DWIN32 /D_WINDOWS /W3 /GR /EHsc
> -- Python3 interpreter:D:/IfcOpenShell-master/deps-vs2017-x86-installed/Python34/python.exe
> -- Python include directory:
> -- Python library release: D:/IfcOpenShell-master/deps-vs2017-x86-installed/Python34/libs/python34.lib
> -- Disabled SWIG warnings
> -- Disabled deprecation warnings for oce
> -- pythonOCC modules will be installed to: D:/IfcOpenShell-master/deps-vs2017-x86-installed/Python34/Lib/site-packages/OCC
> -- Found MSVC compiler: 1 1916
> -- Configuring done
> -- Generating done
> -- Build files have been written to: D:/IfcOpenShell-master/deps-vs2017-x86-installed/Python34/Lib/pythonocc-core

I got lots of VC++ Project Filters in the pythonocc-core folder. And then, I cannot find how to make the command line "make and make install" work properly (I launch cmd with administrator rights) but get always the same message: 

`D:mypath\pythonocc-core> make`
> mingw32-make.exe: *** No targets specified and no makefile found.  Stop.

I also saw that this Cmake File exists: `cmake_install` but there is actually no Makefile that could be a target for the make line. Do you have any idea what went wrong and how to solve that? Any advice would be very helpful, thank you for your attention.

tpaviot, 2020-03-03:
see https://github.com/tpaviot/pythonocc-core/blob/master/ci/conda/bld.bat , it's the way pythonocc-core is built on windows, using ninja

tpaviot, 2020-03-03:
The installation.md may not be up to date, sorry

sohertz, 2020-03-03:
Thank you for your reply. So I installed ninja and used the command line in the cmd prompt. I got following message: 

>  The C compiler identification is GNU 4.5.2
> The CXX compiler identification is GNU 4.5.2
> Check for working C compiler: C:/MinGW/bin/gcc.exe
> Check for working C compiler: C:/MinGW/bin/gcc.exe -- works
> Detecting C compiler ABI info
>  Detecting C compiler ABI info - done
> Detecting C compile features
> Detecting C compile features - done
>  Check for working CXX compiler: C:/MinGW/bin/c++.exe
> Check for working CXX compiler: C:/MinGW/bin/c++.exe -- works
> Detecting CXX compiler ABI info
> Detecting CXX compiler ABI info - done
> Detecting CXX compile features
> Detecting CXX compile features - done
> 
> Found OpenGL: opengl32
> **CMake Error at C:/Program Files/CMake/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
>   Could NOT find Python3 (missing: Python3_EXECUTABLE Interpreter) (found
>   version "3.7")**
> Call Stack (most recent call first):
>   C:/Program Files/CMake/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
>   C:/Program Files/CMake/share/cmake-3.16/Modules/FindPython/Support.cmake:2214 (find_package_handle_standard_args)
>   C:/Program Files/CMake/share/cmake-3.16/Modules/FindPython3.cmake:300 (include)
>   CMakeLists.txt:99 (find_package)
> 
> 
>  Configuring incomplete, errors occurred!
> See also "D:/IfcOpenShell-master/deps-vs2017-x86-installed/Python34/Lib/pythonocc-core/build/CMakeFiles/CMakeOutput.log".

Do you know how I could fix this? Is it possible to add some arguments in the request?

chetanjain2099, 2020-04-13:
> Thank you for your reply. So I installed ninja and used the command line in the cmd prompt. I got following message:
>
>
> Do you know how I could fix this? Is it possible to add some arguments in the [...]

I have the same issue. Can someone tell me how to fix this?

htlcnn, 2020-04-18:
I had the same issue with Python 3.6. Could not find Python3 when I installed Python 3.6 to C:\Program Files\Python3