tpaviot/pythonocc-core #787: PythonOCC access to second constructors.

KeithSloan, 2020-03-08:
If I look at [https://www.opencascade.com/doc/occt-6.9.1/overview/html/occt_user_guides__modeling_algos.html#occt_modalg_4_1_3](url) for example
BRepPrimAPI_MakeCylinder has two constructors in C++ but Python only allows one constructor as far as I know. So How can I make the equivalent of the second constructor via Python? i.e. with an angle

tpaviot, 2020-03-09:
I don't see any issue:

```python
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeCylinder

# 1st way to build the cylinder, using radius and height
radius=10
height = 50
c1 = BRepPrimAPI_MakeCylinder(radius, height)

# 2nd way, using radius, height, and angle
myR = 10
myH = 50
myAngle = 0.1
c2 = BRepPrimAPI_MakeCylinder(myR, myH, myAngle)

# have a look at the __init__ method docstring to see all availabe constructors
help(BRepPrimAPI_MakeCylinder.__init__)
```
There's indeed only one ```__init__``` method, but this unique method gathers all possible C++ constructors, according to the number of arguments.

KeithSloan, 2020-03-10:
Yes I had missed from the documentation that the angle was a default parameter

tpaviot, 2020-03-10:
Well, it's not really a default parameter, from a python perspective, i.e. the python ```___init__``` method for ```BRepPrimAPI_MakeCylinder``` is not
```python
def __init__(self, R, H, angle=2*pi)
```

closed 2020-03-12 by tpaviot