tpaviot/pythonocc-core #788: how to read a step file and convert it to stl file?

BoshraKrout, 2020-03-11:
i've been trying this code to read a step file and convert it to stl file but it's **crashing** at print('here1')

`from OCC.Core.STEPControl import STEPControl_Reader
from OCC.Core.StlAPI import StlAPI_Writer

input_file  = 'ap242.stp'   # input STEP (AP203/AP214 file)

print('here')
step_reader = STEPControl_Reader()
step_reader.ReadFile( input_file )
print('here1')
step_reader.TransferRoot(1)
print('here2')
myshape = step_reader.Shape()
print("File readed")

stl_writer = StlAPI_Writer()
stl_writer.SetASCIIMode(True)
stl_writer.Write(myshape, 'ap242.stl')
print("Written")`

please help me! i need this so bad :(

tpaviot, 2020-03-11:
can u please upload the step file as well

BoshraKrout, 2020-03-11:
[ap242.zip](https://github.com/tpaviot/pythonocc-core/files/4318678/ap242.zip)

tpaviot, 2020-03-11:
it seems your STEP file can't be read:

```python
from OCC.Core.STEPControl import STEPControl_Reader

input_file  = 'ap242.stp' 

step_reader = STEPControl_Reader()
status = step_reader.ReadFile(input_file)

# status should be either
# IFSelect_RetVoid = 0,
# IFSelect_RetDone = 1
# IFSelect_RetError = 2
# IFSelect_RetFail = 3
# IFSelect_RetStop = 4
print(status)
```
It outputs ```3```, that is to say ```IFSelect_RetFail```. It comes either from the STEP file itself (no compliant with the standard) or from the OCCT kernel and this should be reported upstream.

tpaviot, 2020-03-11:
In both cases, it does not come from the python wrapper, I close the issue please report any feedback anyway

closed 2020-03-11 by tpaviot

BoshraKrout, 2020-03-12:
the issue was from the STEP file it self, it worked when i changed it.
thanx