tpaviot/pythonocc-core #789: Static Type Hints with python 3.6

BloodWorkXGaming, 2020-03-11:
First of all, very great library with a great amount of good examples.

However working with it is a bit painful. 
As a newcomer to this, i have to check to check the docs quite often, as I am unable to rely on the PycharmSuggestions as there are no hints given.
According to pycharm, every method returns itself.

My way of dealing with it is to add types to the variables, but is it only a very slow and annoying workaround:
```
location: TopLoc_Location = solid.Location()
        transform: gp_Trsf = location.Transformation()
        translation: gp_XYZ = transform.TranslationPart()
        rot: gp_Quaternion = transform.GetRotation()
```

Adding the types directly into the library will benefit many other users und reduce the risk of running into bugs on execuiton.

Thank you :)

tpaviot, 2020-03-12:
I fully agree with you, that's something I started to think about  a couple of weeks ago.

tpaviot, 2020-03-12:
It would also enable static type checking using mypy for example

jf---, 2020-03-12:
Yes yes and yes... 
The angle of attack is through `pythonocc-generator` I figure?
This would be a useful way to contribute to that aspect of pythonocc, I'd be happy to take a stab at this...

tpaviot, 2020-03-13:
I'm quite new to py3 type hints, never used it, neither with mypy or Pycharm.  @jf--- yes the pythonocc-generator is the way to go. I did not see any easy solution to make SWIG generate type hints. And if ever it was the case, types passed to/return by the wrapper over occt is close to c++, (e.g ```opencascade::handle<SomeClass>```, and would require a modification to be more explicit.

In my opinion, the fastest path would be to export type hints, using the generator, to what is known as stub file(s), that is to say some ```.pyi``` files that only contain type hints. See https://mypy.readthedocs.io/en/stable/stubs.html or https://www.jetbrains.com/help/pycharm/stubs.html#

There are some tools to automatically generate stub files from python modules or packages (https://mypy.readthedocs.io/en/stable/stubgen.html), but doing it from the pythonocc-generator would allow more control over what is performed.

I wonder wether we should have to generate big ```OCC.pyi``` stub files or create one ```pyi``` stub file per module (```gp.pyi```, ```BRepPrimAPI.pyi``` etc).

tpaviot, 2020-03-13:
Note for myself, a useful resource here https://github.com/python/typeshed , type hints for python standard modules

tpaviot, 2020-03-13:
@BloodWorkXGaming @jf--- attached a first draft for a stub pyi file for the gp module. it was automatically generated by a modified pythonocc generator. Can you tell me if you manage to get something with it ?

[gp.pyi.zip](https://github.com/tpaviot/pythonocc-core/files/4328078/gp.pyi.zip)

jf---, 2020-03-13:
> I'm quite new to py3 type hints, never used it, neither with mypy or Pycharm

Not an expert in type theory - quite far from it, but do have been using typehints a lot the past 2 yrs or so... 

> in my opinion, the fastest path would be to export type hints, using the generator, to what is known as stub file

Hey, that's an excellent idea - didnt think of this, but that's definitely a very practical idea...

> attached a first draft for a stub pyi file for the gp module. it was automatically generated by a modified pythonocc generator

unbelievable... that was near-instant??? 
duly impressive --- moreover good news:

see how `pycharm` correctly deduces that `gp_Ax1().Direction()` is a `gp_Dir` type?

[image]

This can't be deduced from the `SWIG` python code:

[image]

What more can I say --- this is a **huge** step fwd in my book and enough motivation for a `7.4.1` release...

@tpaviot I salute you, incredible!

[image]

tpaviot, 2020-03-13:
@jf--- thank you for this feedback. I just *love* the video ! Rasta rocket, right ?

Good that this first shot works as expected. Just out of curiosity, where did you put the gp.pyi file ? and did you configure pycharm to use the stub file or did he find it himself ?

Modifying the generator to produce pyi files is not that difficult, the code base allows it without major difficulty. Still some fixes to commit and I'll share it

tpaviot, 2020-03-13:
Ok, I downloaded/instll PyCharm community. Actually copying the pyi file at the same level than the python module .py is enough, that's pretty simple.

jf---, 2020-03-13:
> I just love the video ! 
😎
> Actually copying the pyi file at the same level than the python module .py is enough, that's pretty simple

Yes, amazing that you got it right, right out of the box... `pycharm` is quite incredible...

***

So -- what about porting the non-generated `pythonocc` code.
I'm up for taking this on and there are also ample tools to facilitate the process ( see [this talk](https://github.com/Python-Meetup-Rotterdam/meetup3/blob/master/typehints/typehints.md) ).

That said, while I think that the `.pyi` approach is ideal for generated code, for written code, its a different call. There are subtle differences in the way type-hinting works across versions. Hence, this leads to discussion wrt sunsetting old python versions.

Since this is 2020, with 3.8 released and 3.9 coming in March, I'd be rather in favor of supporting python 3.6+

Can I have your input on that @tpaviot ?

rainman110, 2020-03-13:
Hi guys. @tpaviot awesome work, congrats. 🎉 This is definitely a big step forward.

When I saw this issue initially, my thought was that it is not possible since SWIG does not yet support this. Great, that you proved me wrong ;)

jf---, 2020-03-13:
> was that it is not possible since SWIG does not yet support this

My guess is that the `.pyi` file is generated through `pythonocc-generator`, rather than `SWIG`?

jf---, 2020-03-13:
An observation wrt an object ( `gp_Ax1` ) that has `typehints` is that it **is** a properly documented method ( see last image ), but its not showing up in the IDE / ipython / typing `help(x.Direction)`. Do I recall correctly that earlier version did have documented methods?

[image]

[image]

[image]

[image]

rainman110, 2020-03-13:
Yes, it is created via the generator. I actually did not know, that it is possible to create a .pyi file. The alternative would have been to modify the SWIG-generated code, which seemed to be a crazy idea for me. 😄

jf---, 2020-03-13:
@tpaviot a small detail is that it seems that `constants` are missing in the generated `.pyi` file

[image]

tpaviot, 2020-03-13:
@rainman110 yes, SWIG seems to not be good as type hinting. I pushed code to github, see https://github.com/tpaviot/pythonocc-generator/tree/review/stub-pyi

@jf--- I've been working on the documentation. I tried to automate the sphinx documentation at readthedocs, but if fails to complete because it takes too much time. FYI, Irecently changed the documentation from rst to numpydoc, which is much easier to read. I experimented different SWIG options, currently docstrings are embeded into the c++ wrapper, not into the python interface, which results in smaller python modules

tpaviot, 2020-03-13:
@jf--- @rainman110 please forgive the poor code quality, I started this work this morning, trying to go as fast as possible

tpaviot, 2020-03-13:
@jf--- yes, constants are missing, they come from c++ named enums, I have to take this into account

tpaviot, 2020-03-13:
@jf--- in your example, ```gp_XOY``` is not a constant, it's the wrapper for the static method ```XOY``` of the ```gp``` class (there's a ```gp``` class in the ```gp``` module indeed).

Try
```python
from OCC.Core.gp import gp
gp.XOY()
```
that should work. But indeed ```gp_XOY``` should be added to the type hint stub file as well, I have to think about how to

tpaviot, 2020-03-24:
All .pyi files were updated and upload to the branch review/update-wrapper, related commits ab439efec7160cb34e4275c53d2bf40f94284117 and 6ffa10ce9c25ddccec6e14ddc5544fa58b82d40c

jf---, 2020-03-24:
Sorry for the delays -- let me take a look at this in a few hrs... wonderful @tpaviot !

BloodWorkXGaming, 2020-03-26:
Awesome work so far, thanks a lot!
A few things I noticed so far: 
- Specifically the method 
```
rot: gp_Quaternion = transform.GetRotation()
(ex, ey, ez) = rot.GetEulerAngles(gp_Intrinsic_XYZ)
``` 
ist working in python, however the type are saying a wrong thing:
`def GetEulerAngles(self, theOrder: gp_EulerSequence, theAlpha: float, theBeta: float, theGamma: float) -> None: ...` as if there are no returns (prob due to pointers in c++)

- The main code has now typehints, but the custom Extend and Display packages are still lacking them. Here they could probably added directly into the code

- `gp_Intrinsic_XYZ` is not being shown as a valid import (still works tho)

tpaviot, 2020-03-26:
@BloodWorkXGaming thank you for this detailed feedback

* you're right wrt the ```GetEulerAngles``` method, and other similar. The c++ function prototype returns float by reference, which requires to be adapted to python. The documentation/type hint does not reflect the change in the Python api.

* yes,  Extend and Display packages are pure python so type hints can be added directly to the module code. Had no time to do it yet, @jf--- wanted to help, it could be a great contribution (docstring clean up updates + type hints)

* In the branch ```review/update-wrapper```, I modified the python wrapper so that c++ enums can be accessed by their full names. For instance,

```bash
>>> from OCC.Core.gp import *
>>> gp_EulerSequence.gp_Intrinsic_XYZ  # this is new in dev master branch
8
>>> gp_Intrinsic_XYZ  # this used to work in 7.4.0
8
```

Attached an example in PyCharm, it's not what you need ?

[image]

jf---, 2020-03-26:
Looks like it can be merged indeed. @tpaviot yes that's a good suggestion, I'll explore looking into the docstrings

tpaviot, 2020-03-27:
@jf--- there are many files to process, it's a huge work. I suggest we open a new issue so that  I can see which module you're working on, so that I choose another one and we move in a ```multiprocess``` mode

tpaviot, 2020-03-27:
@jf--- there is still a few work to achieve before a PR can be opened, see for instance 3d5913e1