tpaviot/pythonocc-core #790: DLL Load Error on Windows

servo270, 2020-03-13:
I have been trying to get pythonocc to work recently, but have been running into the following error:
[image]

Sytem information: Windows 10 Pro Education
Python Version: installed up to 3.7, using 3.6 for environment
PythonOCC version: 7.4.0
Installed by: conda install -c dlr-sc pythonocc-core 
Relavent conda environment setup:
[image]
System user path:
[image]

To trigger the error, the following was run:
`from OCC.Visualization import Display3d`

rainman110, 2020-03-13:
Hi @servo270! Could you please print also the output of ```conda list```.

servo270, 2020-03-13:
@rainman110 attached is the output (it's too long to paste here)
[conda list.txt](https://github.com/tpaviot/pythonocc-core/files/4329280/conda.list.txt)

rainman110, 2020-03-13:
Wow... these are a lot of packages. Hard to figure out, what the cause of the problem is.

I see, you are still using pythonocc 0.17.3 (not 7.4.0). This is okay, as TiGL does not yet support a newer version yet.

The first step I would do in your case is to create a clean conda environment:
conda create -n occtest python=3.6 pythonocc-core=0.17.3
conda activate occtest

Then try again import the Visualization package. If this works, try adding other packages one after another. This way, we could figure out, where the DLL collision comes from.

servo270, 2020-03-16:
Unclear what caused the original collision problem, but manually reinstalling the packages starting with:
conda create -n envname python=3.6 pythonocc-core=0.17.3

eliminated the issue

closed 2020-03-16 by servo270