tpaviot/pythonocc-core #791: Message_ProgressIndicator not found

e3dEF, 2020-03-13:
Hi,
I want to set a Message_ProgressIndicator in a BOPAlgo_Builder function using the current conda build, but I can't import the Progress Indicator. Can you help me please?

```
from OCC.Core.BOPAlgo import BOPAlgo_Builder,
from OCC.Core.Message import Message_ProgressIndicator <- not found
builder = BOPAlgo_Builder()
indicator = ???
builder.SetProgressIndicator(indicator) # 

void BOPAlgo_Options::SetProgressIndicator | ( | const Handle< Message_ProgressIndicator > & | theObj | )
```
[https://www.opencascade.com/doc/occt-7.4.0/refman/html/class_b_o_p_algo___options.html#a68a5cdf32404e014c4ca70093f3f7359](url)
[https://www.opencascade.com/doc/occt-7.4.0/refman/html/class_message___progress_indicator.html](url)

tpaviot, 2020-03-15:
So far, the ```Message``` module is empty, the python wrapper does not provide any access to ```Message_*``` classes. Leave this issue opened and keep posted.

rainman110, 2020-03-15:
Hi guys. I know this class only from c++, where it is an interface class actually. It is meant to implement a custom progress indicator and hand it over to the algorithm.

If we want to support this in pythonocc, we have to enable the directors feature of swig. Then you can use polymorphism over language borders.

tpaviot, 2020-03-16:
@rainman110 do you have an example of using it from c++ ? something simple that could be ported to python for a test ?

geraldozzz, 2020-12-24:
Hi,

I also tried to use the Message_ProgressIndicator in breptools, but got error like below, can you please help? @tpaviot 
```
Traceback (most recent call last):
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2020.3\plugins\python-ce\helpers\pydev\_pydev_bundle\pydev_monkey_qt.py", line 192, in _new_run
    return self._original_run()
  File "D:\mz_dev\dev\svwpoc\model_io.py", line 118, in run
    breptools_Read(model, self.filename, builder, pi)
  File "D:\mz_dev\python\envs\svwpoc\lib\site-packages\OCC\Core\BRepTools.py", line 934, in breptools_Read
    return _BRepTools.breptools_Read(*args)
TypeError: in method 'breptools_Read', argument 4 of type 'opencascade::handle< Message_ProgressIndicator > const &'
```
Code:
```
class ProgressIndicator(Message_ProgressIndicator):
    def __init__(self, parent, min_value, max_value):
        progress = QProgressDialog('load cad', 'loading', min_value, max_value, parent=parent)
        self.myProgress = progress

    def Show(self):
        name = self.GetScope(1).GetName()
        progress = self.myProgress
        if name:
            progress.setLabelText(name)
        p = self.GetPosition()
        value = progress.minimum() + p * (progress.maximum() - progress.minimum())
        progress.setValue(value)
        QApplication.processEvents()

    def UserBreak(self):
        return self.myProgress.wasCanceled()

model = TopoDS_Shape()
builder = BRep_Builder()
pi = ProgressIndicator(self.parent, 0, 100)
breptools_Read(model, self.filename, builder, pi)
```

NoNeedToHurry, 2021-03-07:
Hello, 
Maybe the examples in this post could help with version 7.5.0 :
https://unlimited3d.wordpress.com/2020/10/17/progress-indication-changes-in-occt-7-5-0/