tpaviot/pythonocc-core #792: The use of camera

gborangecat, 2020-03-14:
I am working on a project about 6D object pose estimation, in which I need to generate a couple of CAD model templates from different perspectives, e.g. taking samples by placing camera in 0~180° yaw/pitch angle around and 10mm~1000mm distance from the model. By capturing the model templates I can proceed my project to template matching, and then use the information of templates to estimate the real pose of object in real-world camera.
Now I have already had the stp file of the object, and I can read it and display it in pythonocc. I searched for tutorials about pythonocc but I cannot find a detailed description to manipulate the camera.  The only related example I found is in "pythonocc-demos-master/pythonocc-demos-master/examples/core_visualization_camera.py". 

`from OCC.Display.SimpleGui import init_display
from OCC.Extend.DataExchange import read_step_file

display, start_display, add_menu, add_function_to_menu = init_display()

# load step file
the_shape = read_step_file('../assets/models/as1_pe_203.stp')
display.DisplayShape(the_shape)

def animate_viewpoint():
    display.FitAll()
    display.Context.UpdateCurrentViewer()

    cam = display.View.Camera()  # type: Graphic3d_Camera

    center = cam.Center()
    eye = cam.Eye()

    for i in range(100):
        eye.SetY(eye.Y() + i)
        cam.SetEye(eye)
        display.View.ZFitAll()
        display.Context.UpdateCurrentViewer()

    for i in range(100):
        center.SetZ(center.Z() + i)
        cam.SetCenter(center)
        display.View.ZFitAll()
        display.Context.UpdateCurrentViewer()

add_menu("camera")
add_function_to_menu("camera", animate_viewpoint)

start_display()`

I got confused by the code. So I would like to raise an issue about this question. **Can anyone show me how to manipulate the camera in pythonocc?**

nsmela, 2023-10-12:
Sorry to revive an old thread, but felt it was better than creating another with the same topic.

I am also looking to manipulate a camera in pythonocc. I want to make an orbit camera:

1. Camera is always looking at 0,0,0
2. Right Mouse Drag X rotates camera around the y axis in the scene
3. Right Mouse Drag Y raises or lowers the tilt or yaw of the camera, but to never exceed 90 deg or go below -90 deg
4. Mouse Wheel zooms in or out
5. Left Mouse to select or draw a rect to select multiple items

My approach is to make a new [qtViewer](https://github.com/tpaviot/pythonocc-core/blob/a43f1f7d0e27bdfd427e245c809dcf1d74c77a4c/src/Display/qtDisplay.py#L63) class redefining the mousePressEvent, mouseMoveEvent, and mouseReleaseEvent functions. I can work out the math for it later, but I'm struggling to find the right method to rotate around a specific axis, not the view's up axis.

I think the info is on OpenCascade's codebase, specifically: [v3d_view](https://dev.opencascade.org/doc/refman/html/class_v3d___view.html) & [Rotate](https://dev.opencascade.org/doc/refman/html/class_v3d___view.html#a9e58690bd32c076c96df19fbc28bacb5)
The issue is I don't know how to reference the [V3d_TypeOfAxe](https://dev.opencascade.org/doc/refman/html/_v3d___type_of_axe_8hxx.html#a36c137b2c17047b84d1e2de451a0061a) listed as a possible parameter. I've found it under [src/SWIG_files/wrapper/V3d.pyi](https://github.com/tpaviot/pythonocc-core/blob/c2f9d1b12a1fbc5d27f668a1753a0945e6d57a7b/src/SWIG_files/wrapper/V3d.pyi#L68C1-L68C1)

So, am I on the right track? Am I right that if I can somehow use the v3d_TypeOfAxe_->V3d_Y I could rotate around the upwards axis? If so, how so I turn [self._display_View.Turn()](https://github.com/tpaviot/pythonocc-core/blob/a43f1f7d0e27bdfd427e245c809dcf1d74c77a4c/src/Display/qtDisplay.py#L239) in this way?

tpaviot, 2023-10-13:
I don't know about the `V3d_TypeOfAxe` param of the camera, I've never had to use it, same for the `Turn` method.

Note that the rotation is defined in the OCCViewer class, see https://github.com/tpaviot/pythonocc-core/blob/master/src/Display/OCCViewer.py#L679

nsmela, 2023-10-13:
Yes, but I think that's based off the view's position in the scene. I wanted to rotate around an axis regardless of that.

I'm having limited success with using SetEye by using mouse y change the z setting. It's adding the tilt I was looking for. But I can't seem to combine it with the mouse x