tpaviot/pythonocc-core #793: Problem importing from OCC.Extend and OCC.Display

avansisintegracion, 2020-03-18:
Hi there! I am working with python 3.7 and pythonocc-core 7.4.0 and I am trying to import these methods:
- from OCC.Extend.DataExchange import write_step_file
- from OCC.Display.SimpleGui import init_display 
The thing is that I can import anything that comes from OCC.Core but not from OCC.Extend or OCC.Display and I don't know if I am doing something wrong.

The error when I am trying to import the method write_step_file is the following one:

Traceback (most recent call last):
  File "/mnt/d/Documents/workspace/prueba/step_to_stl_prueba.py", line 10, in <module>
    from OCC.Extend.DataExchange import write_step_file
  File "/root/miniconda3/envs/condaPrueba/lib/python3.7/site-packages/OCC/Extend/DataExchange.py", line 32, in <module>
    from OCC.Core.XCAFDoc import (XCAFDoc_DocumentTool_ShapeTool,
  File "/root/miniconda3/envs/condaPrueba/lib/python3.7/site-packages/OCC/Core/XCAFDoc.py", line 18, in <module>
    from . import _XCAFDoc
ImportError: libGL.so.1: cannot open shared object file: No such file or directory

And the error when I try to import the method init_display from SimpleGui is the next one:

_Traceback (most recent call last):
  File "/mnt/d/Documents/workspace/prueba/coreLoadIges.py", line 7, in <module>
    from OCC.Display.SimpleGui import init_display
  File "/root/miniconda3/envs/condaPrueba/lib/python3.7/site-packages/OCC/Display/SimpleGui.py", line 26, in <module>
    from OCC.Display.OCCViewer import OffscreenRenderer
  File "/root/miniconda3/envs/condaPrueba/lib/python3.7/site-packages/OCC/Display/OCCViewer.py", line 28, in <module>
    from OCC.Core.Aspect import Aspect_GFM_VER
  File "/root/miniconda3/envs/condaPrueba/lib/python3.7/site-packages/OCC/Core/Aspect.py", line 18, in <module>
    from . import _Aspect
ImportError: libGL.so.1: cannot open shared object file: No such file or directory_
Thank you for your help!

jf---, 2020-03-19:
This looks like an installation issue. Can you try to create another conda environment?

tpaviot, 2020-03-25:
It seems that your OpenGL installation is wrong. What's your machine configuration ? os ? version ?

lokit-iscitech, 2021-08-10:
I am getting the same error:

```
Traceback (most recent call last):
  File "c:/Users/91917/Desktop/OpenCASCADE/main.py", line 9, in <module>
    import OCC.Display.qtDisplay as qtDisplay
  File "E:\ProgramData\Anaconda3\envs\occenv1\lib\site-packages\OCC\Display\qtDisplay.py", line 25, in <module>
    from OCC.Display import OCCViewer
  File "E:\ProgramData\Anaconda3\envs\occenv1\lib\site-packages\OCC\Display\OCCViewer.py", line 27, in <module>
    from OCC.Core.Aspect import Aspect_GFM_VER
  File "E:\ProgramData\Anaconda3\envs\occenv1\lib\site-packages\OCC\Core\Aspect.py", line 18, in <module>
    from . import _Aspect
ImportError: DLL load failed while importing _Aspect: The specified module could not be found.
```
This is one the fresh environment. My OS is Win10 with Intel HD graphics 620. Version of pythonocc-core is 7.5.1.

moker-spaghetti, 2023-12-05:
My conda env in windows, python 3.10, pythonocc-core==7.5.1 caused the same problem.
Installing `libdeflate` and renaming `env/Library/bin/deflate.dll` to `libdeflate.dll` solved it in my case.
Cheking DLL dependencies by `dependencies` hepled a lot.

Nadjei74, 2024-04-04:
Hi, I tried the PythonOcc beginners example from the analysis Situs forum but I keep encountering this errror
[image]
Can someone please help

Bill-XU, 2024-08-11:
> Hi, I tried the PythonOcc beginners example from the analysis Situs forum but I keep encountering this errror ![PythonOcc issue](https://private-user-images.githubusercontent.com/165973434/319412687-06132faf-7d14-4d64-ad4f-5c8582729643.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMzNTY3NzcsIm5iZiI6MTcyMzM1NjQ3NywicGF0aCI6Ii8xNjU5NzM0MzQvMzE5NDEyNjg3LTA2MTMyZmFmLTdkMTQtNGQ2NC1hZDRmLTVjODU4MjcyOTY0My5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgxMVQwNjA3NTdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMDg3NDJiN2Q2NDU1ZjI4MTAyNTgwZTliZjhiNWQ5YTBhZGRhZGRkNWRhZjAxMzA5ZDJjMTJjOWY3OTFjZDM5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.IxoWemFtAJ_hAkX8hEGTp_ywr7Dnkpg9rXZhwNP9H80) Can someone please help

have  you installed pythonocc?