tpaviot/pythonocc-core #795: OCCT 7.4.0: Undefined references, again

Andrei-Pozolotin, 2020-03-20:
1. while trying to build:
https://github.com/CadQuery/cadquery/issues/226
https://github.com/CadQuery/cadquery

2. with pythonocc-core release from conda-forge/pythonocc-core
https://github.com/tpaviot/pythonocc-core/releases/tag/7.4.0

3. in an experimental fork:
https://github.com/random-builder/design_cadquery/commit/8a8a09800951cb9faf599c59b39d1599f4313067

4. we run into issue similar to:
https://github.com/tpaviot/pythonocc-core/issues/709 OCCT 7.4.0: Undefined references when loading modules

5. with build log:
[tool/build-all.txt](https://github.com/random-builder/design_cadquery/blob/8a8a09800951cb9faf599c59b39d1599f4313067/tool/build-all.txt)

6. reporting error:
```
E   ImportError: $PREFIX/lib/python3.7/site-packages/OCC/Core/_BRepFilletAPI.so: undefined symbol: _ZNK20BRepBlend_AppSurface10TolReachedERdS0_
```
7. so ... who/where is expected to fix this? :-)

tpaviot, 2020-03-20:
The problem occurs when building a third part software. Can you please first check with the related team. If ever the issue comes from pythonocc itself, can you please post a way to reproduce the issue

tpaviot, 2020-03-20:
I had a look at the wrapper interface file, the method ```TolReached``` of the class ```BRepBlend_AppSurface``` is actually explicitly removed from the methods to process, see https://github.com/tpaviot/pythonocc-generator/blob/master/src/Modules.py#L584

tpaviot, 2020-03-20:
ok, I got it. The pythonocc-core package from conda-forge channel is corrupted:
```
(7.4.0-cf) thomas@thomas-X756UVK:~/$ python
Python 3.7.6 | packaged by conda-forge | (default, Mar  5 2020, 15:27:18) 
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from OCC.Core.BRepBlend import *
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/thomas/miniconda3/envs/7.4.0-cf/lib/python3.7/site-packages/OCC/Core/BRepBlend.py", line 18, in <module>
    from . import _BRepBlend
ImportError: /home/thomas/miniconda3/envs/7.4.0-cf/lib/python3.7/site-packages/OCC/Core/_BRepBlend.so: undefined symbol: _ZNK20BRepBlend_AppSurface10TolReachedERdS0_
```
whereas dlr-sc package is not
```
(7.4.0-dlrsc) thomas@thomas-X756UVK:/$ python
Python 3.7.6 | packaged by conda-forge | (default, Jan  7 2020, 22:33:48) 
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from OCC.Core.BRepBlend import *
>>> 
```

This might has to do with occt compilation itself. @rainman110  @looooo any idea ?

rainman110, 2020-03-20:
Yes, its a bug in OCCT. I made a patch for this on the DLR repo:
https://github.com/DLR-SC/tigl-conda/blob/master/opencascade/fix_brepblend.patch

tpaviot, 2020-03-20:
Ah, interesting. @looooo  is it possible to include this patch as well in the occt conda-forge package ?

In the meantime @Andrei-Pozolotin I guess you can switch to the dlr-sc pythonocc-core package

looooo, 2020-03-20:
https://github.com/conda-forge/occt-feedstock/pull/37

adam-urbanczyk, 2020-03-20:
@rainman110 isn't this a bug in the bindings? The function is `inline` in OCCT and it seems PythonOCC is trying to linke against it.

rainman110, 2020-03-20:
@adam-urbanczyk No, believe me. I tried to deactivate it in the bindings without success.

Inlining functions in *.cpp files is not a good idea. In the header, it would have been okay though. Here, the problem is, that another function from a header file references the this function, but the code is never compiled (at least not in the wrapper and just inlined in OCCT). So there is no way to bind this function. Another alternative would have been to completely skip the BRepBlend.hxx header file.

This supports my findings: https://tracker.dev.opencascade.org/view.php?id=31020

Andrei-Pozolotin, 2020-03-20:
thank you, guys.

1. I can confirm that switch to `dlr-sc` does enable general project build
https://github.com/random-builder/design_cadquery/commit/d0f31155078590aa890d0d77e09f50fb2a3d8278

2. I still face some intractable "conda problem", build log reports that `__cuda` is bad :-)
[tool/build-all.txt](https://github.com/random-builder/design_cadquery/blob/d0f31155078590aa890d0d77e09f50fb2a3d8278/tool/build-all.txt)
```
failed to get install actions, retrying.  exception was: Unsatisfiable dependencies for platform linux-64: {'feature:/linux-64::__cuda==10.2=0', 'feature:|@/linux-64::__cuda==10.2=0'}
```
perhaps your conda experience can point me in the right direction (i.e. disable irrelevant dependency)?

tpaviot, 2020-03-20:
I guess there are conflicts between dependencies.

@Andrei-Pozolotin @looooo just updated the occt conda-forge package to include @rainman110 's patch. I suggest you upgrade to the latest build and report if it's ok.

adam-urbanczyk, 2020-03-20:
@rainman110  you are absolutely right! Thanks for the explanation.

Andrei-Pozolotin, 2020-03-20:
update report:

1. I confirm that now switch back to `conda-forge` 
brings dependency to `linux-64/occt-7.4.0-h9121d39_3.tar.bz2`
which allows `cadquery` to build fine (allows to get to meaningful `pytest` errors)
https://github.com/random-builder/design_cadquery/commit/a42a7eeb82ba72e735852a3e6c0bf0182c34269f

2. this issue is now resolved, thanks again to all

closed 2020-03-21 by tpaviot