tpaviot/pythonocc-core #798: WebGl pointcloud

matthijsramlab, 2020-03-23:
Would it be possible to view a point cloud with the WebGL renderer? At the moment it only takes a TopoDS_Shape as an argument

jf---, 2020-03-24:
hi @matthijsramlab 

https://github.com/tpaviot/pythonocc-demos/blob/292af90d1e75b57bef23faf8fc223c0963537722/examples/core_display_point_cloud.py#L55-L67

tpaviot, 2020-03-25:
@matthijsramlab It's already possible in the JupyterRenderer, see https://github.com/tpaviot/pythonocc-demos/blob/master/jupyter_notebooks/point_cloud.ipynb but I didnot implement this feature into the w3dom or threejs renderer. Isn't the jupyter renderer enough for your needs ?

matthijsramlab, 2020-03-30:
@tpaviot and @jf--- thanks for the answers. The Jupyter notebook example is very nice. However I would like to render it in my Flask app. So if that would be possible, it would be awesome.

jf---, 2020-03-31:
> So if that would be possible, it would be awesome

Sure it is,  but it'll be a part of your app, not pythonocc...
What's missing from the [`core_webgl_*`](https://github.com/tpaviot/pythonocc-demos/tree/master/examples) examples that keeps you from taking the leap?

Wrt `flask`, its straightfwd these days to make an app of a notebook via [voilà](https://github.com/voila-dashboards), highly recommended

tpaviot, 2020-04-01:
@matthijsramlab the "point cloud" feature implemented in the jupyter renderer is missing from the Webgl renderer, right. It has to be added, not a big effort though.

RobinCPC, 2020-04-01:
Hi @tpaviot and @jf---, I add the "point cloud" feature in Webgl renderer of my fork.
Actually, I add a flask server for the WebGL renderer.
Here is the commit
https://github.com/RobinCPC/pythonocc-core/commit/dc367c9483c1987a7995fb52962046d3cb9390ea

Hi @matthijsramlab, 

Here is the result of rendering point cloud.
[image]

This is the sample code I used:
https://gist.github.com/RobinCPC/cdbb40001fb5817625cf4e91d87ab4ea

tpaviot, 2020-04-01:
@RobinCPC great, using jinja results in a much cleaner code, I will include this work in the pythonocc codebase if you agree

tpaviot, 2020-04-01:
@RobinCPC yes, please open a PR, this is a cool addition

matthijsramlab, 2020-04-01:
@RobinCPC Thanks allot, I will try it out this afternoon

RobinCPC, 2020-04-01:
Hi @tpaviot, Thank you.  I have made a PR. 
Please let me know if anything needs to update.

Hi @matthijsramlab you are welcome.