tpaviot/pythonocc-core #799: No module OCC.Core found

Thmbzh, 2020-03-25:
Hello,
First I want to apologize because I know this question has already been asked but I'm kinda new to coding and programming and this issue is giving me hard times.
For an internship I  have to use PyGeM library. I downloaded it an go through the installation file without any problems (installed Anaconda3, all packages needed). But when I wanted to run the 'test.py' file, I had the error 'ModuleNotFoundError (No module named 'OCC.Core')'
I used the command line to install packages : 
`conda install -c conda-forge -c dlr-sc -c pythonocc -c oce pythonocc-core==0.17 python=3.6`
It has been 2 days I searched online how to fixed it but I'm kinda lost and that's why I'm requesting your help.

tpaviot, 2020-03-25:
This is a very old release. The PyGem team just updated his code base to the latest pythonocc release (7.4.0), see commit https://github.com/mathLab/PyGeM/commit/185c90529a0d67874dc1b52a3940d09356dfd612 two days ago.

IMO you should run the pygem current dev master branch

Thmbzh, 2020-03-25:
Thanks for answering !
I have download the master branch but I don't know, I've followed every step and still get this error.
I tried this command form the file you gave me the address :
conda install --yes -c conda-forge pythonocc-core=7.4.0 numpy scipy matplotlib vtk nose setuptools coveralls python=3.7
And it can't go through (solving environment problems)
I'm not familiar with this and I may doing something wrong but I don't know what.
I'm trully sorry to bother you with this.

tpaviot, 2020-03-25:
@Thmbzh you should post to the pygem issue tracker, they will help you https://github.com/mathLab/PyGeM/issues

Thmbzh, 2020-03-25:
I'll ask them.
Thank you very much for your patience and your time !

tpaviot, 2020-04-04:
@Thmbzh closing the issue, please feel free to report here any answer from the pygem team

closed 2020-04-04 by tpaviot