tpaviot/pythonocc-core #800: Wrong Description on OCC.Extend.DataExchange.read_step_file_with_names_colors

BloodWorkXGaming, 2020-03-26:
In this function the Description is: 

```
    """ Returns list of tuples (topods_shape, label, color)
    Use OCAF.
    """
```

However instead of the label it returns the name. (Which is way less powerfull)
Maybe it makes more sense to change that to returning the label instead of the name to fit the description.

These lines would be affected afaik:

https://github.com/tpaviot/pythonocc-core/blob/master/src/Extend/DataExchange.py#L301

https://github.com/tpaviot/pythonocc-core/blob/master/src/Extend/DataExchange.py#L270

Thanks :)

BloodWorkXGaming, 2020-03-26:
Ooookay, I have no idea why, but I tested it and replaced the name with a label.
Everytime when I tried to iterate over it I got a `exit code -1073741819 (0xC0000005)` so a access violation

Doubt it error is in the python code of this lib

tpaviot, 2020-04-04:
@BloodWorkXGaming can you please post the failing code so that the issue can be identified/fixed

BloodWorkXGaming, 2020-04-13:
Sorry for not answering so long :/

Replacing the two lines liked above to return the label instead of only the name and then iterating over them with eg: 

```
for solid, (label, col) in shape_list.items():
        print(label.GetLabelName())
```

This causes the error, probably due to some cleanup that is happening in c++ even tho there are still referneces to it