tpaviot/pythonocc-core #801: BRepAlgoAPI_Section most functions marked obsolete

KeithSloan, 2020-04-01:
Looking at documentation on BRepAlgoAPI_Section most of the functions are marked at the end of description as obsolete.

Is BRepAlgoAPI_Section no longer supported? Has it been replaced by something else?

tpaviot, 2020-04-01:
You're right, I did not notice that before. The official documentation is quite clear (https://www.opencascade.com/doc/occt-7.4.0/overview/html/occt_dev_guides__upgrade.html), recent changes suggest to actually use BRepAlgoAPI_Section rather than deprecated BRepAlgo_Section (I just CTRL F'ed ```BRepAlgoAPI``` in this html document).

From upstream occt 7.3.0, it's suggested to use BRepAlgoAPI_* instead of BRepAlgo_* (http://git.dev.opencascade.org/gitweb/?p=occt.git;a=commitdiff;h=9ce99357a4d06be37834b475b2b14102b3fe7f74), so I guess this trailing ```Obsolete``` should be removed.

tpaviot, 2020-04-04:
Question replied, closing the issue.

closed 2020-04-04 by tpaviot