tpaviot/pythonocc-core #802: Best  way to determine if a given Shape is 2d or 3d.

KeithSloan, 2020-04-01:
When coding a function that is passed a shape which is either 2d or 3d, what is the best way to determine if its 2d or 3d.

When faced with a similar situation with FreeCAD I have used Volume() which is zero for a flat 2d shape. With PythonOCC should I use the Mass or ShapeType ?

tpaviot, 2020-04-01:
According to this https://github.com/FreeCAD/FreeCAD/commit/1f0bda27f6533494312dbbfec42ff52fbeb55019 commit, I would say it's better to compute shape volume

KeithSloan, 2020-04-01:
Maybe I missed something but have yet to find Volume with PythonOCC. I see Mass, CentreOfMass, matrix_of_inertia.

rainman110, 2020-04-01:
BRepGProp::VolumeProperties (https://www.opencascade.com/doc/occt-7.0.0/refman/html/class_b_rep_g_prop.html#a9c9b04f4d01a22ab6eb19ed05e1a1620)

tpaviot, 2020-04-01:
@KeithSloan You also have a function ready-to-use from the ```ShapeFactory``` module:

```python
from OCC.Extend.ShapeFactory import measure_shape_volume
[... build any TopoDS_Shape, store it in the the_shape variable]
vol = measure_shape_volume(the_shape)
```

tpaviot, 2020-04-01:
and note that the FreeCad team also checks that the shape is closed (```and obj.Shape.isClosed()```)