tpaviot/pythonocc-core #805: Extracting STEP file geometrical entitier

BoshraKrout, 2020-04-01:


tpaviot, 2020-04-04:
see an example at https://github.com/dblanding/kodacad @dblanding

BoshraKrout, 2020-04-10:
thanx a lot :D

rsm-gh, 2020-04-25:
Hello, 

I'm really curious about this because I need it too and I still getting empty names. It is necessary to use some specific version of pythonocc? I made a test compiling the master branch and I can't make it work.

Thank you,
rsm

BoshraKrout, 2020-09-10:
hi @rsm-gh ,
sorry for being late but in case you still need the answer , check your step file maybe its not readable or doesn't have an assembly structure!
i'm using pythonocc-core-7.4.0 and it works!

Subhashplcm, 2024-03-01:
@tpaviot 
Hello Thomas,
I am currently working on a academic project where i have to visualize the .step file. 
I have very little experience in programming, may i know how to visualize the .step file using pythonocc.