tpaviot/pythonocc-core #806: how to point  out which faces updates differs from original faces

liuxin2322, 2020-04-01:
below code shows drilled hole is in object，but how can I know which faces are updated？？ (original object have 3 faces ,new have 4 faces，two faces data  of the old  changs )，it determine what should I transfer  minimum data into webgl !    

appreciated!
``` python
from math import atan, cos, sin, pi

from OCC.Core.BRepAlgoAPI import BRepAlgoAPI_Cut, BRepAlgoAPI_Fuse,BRepAlgoAPI_Common
from OCC.Core.BRepBuilderAPI import (BRepBuilderAPI_Transform, BRepBuilderAPI_MakeWire,
                                BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeFace)
from OCC.Core.BRepFeat import BRepFeat_MakeCylindricalHole
from OCC.Core.BRepPrimAPI import (BRepPrimAPI_MakeSphere, BRepPrimAPI_MakeCylinder,
                             BRepPrimAPI_MakeTorus, BRepPrimAPI_MakeRevol)
from OCC.Core.TColgp import TColgp_Array1OfPnt
from OCC.Core.gp import gp_Ax2, gp_Pnt, gp_Dir, gp_Ax1, gp_Trsf, gp_Vec
from OCC.Display.SimpleGui import init_display

def generate_shape():
    """Create a sphere with faces top and bottom"""
    sphere_radius = 1.0
    sphere_angle = atan(0.5)
    sphere_origin = gp_Ax2(gp_Pnt(0, 0, 0), gp_Dir(0, 0, 1))
    sphere = BRepPrimAPI_MakeSphere(sphere_origin, sphere_radius,
                                    -sphere_angle, sphere_angle).Shape()
    return sphere

def add_feature(base):
    """Add a "feature" to a shape. In this case we drill a hole through it."""
    feature_diameter = 0.8
    feature_origin = gp_Ax1(gp_Pnt(0, 0, 0), gp_Dir(0, 0, 1))
    feature_maker = BRepFeat_MakeCylindricalHole()
    feature_maker.Init(base, feature_origin)
    feature_maker.Build()
    feature_maker.Perform(feature_diameter / 2.0) #PerformBlind()盲孔；PerformThruNext()；PerformUntilEnd()
    shape = feature_maker.Shape()
    return shape

basic_shape = generate_shape()
featured_shape = add_feature(basic_shape)

display,start_display,add_menu,add_function_to_menu = init_display()
display.DisplayShape(featured_shape)
display.FitAll()
start_display()
```

rainman110, 2020-04-01:
Short answer: Use

   ```BRepAlgoAPI_BuilderAlgo.Modified (const TopoDS_Shape &theS)```

https://www.opencascade.com/doc/occt-7.4.0/refman/html/class_b_rep_algo_a_p_i___builder_algo.html#a523cad14bd5e5104e2b683972a4e6bd8

```BrepAlgoAPI_Cut``` inherits from ```BRepAlgoAPI_BuilderAlgo``` so it also has the ```Modified``` method. It allows to figure out, which faces have been changed by the boolean operation.

rainman110, 2020-04-01:
Here's a C++ example how to use this:
https://github.com/DLR-SC/tigl/blob/33debf20979d7d20cd1858fd10516b55d66403e6/src/boolean_operations/CBooleanOperTools.cpp#L37

liuxin2322, 2020-04-01:
@rainman110 very thanks!

liuxin2322, 2020-04-01:
@rainman110 if MakeChamfer/fillet (3d) can also think in this way？？or another handled approach？:grinning:

rainman110, 2020-04-01:
I've never tried, but in principle it should be possible. Filleting also adds new faces, so you might also want to use the method ```Generated```.

liuxin2322, 2020-04-01:
@rainman110 I noticed three functions(generated，modified，deleted),   maybe it can solve my problem! thx:kissing_heart:

closed 2020-04-03 by liuxin2322