tpaviot/pythonocc-core #810: STEP file assembly structure

BoshraKrout, 2020-04-03:
Hello!

I'm working on a step file. i want to see it's assembly structure.
can any one help?

`
from OCC.Display.WebGl import x3dom_renderer
from OCC.Core.STEPControl import STEPControl_Reader

from OCC.Extend.DataExchange import read_step_file_with_names_colors
from OCC.Core.Quantity import Quantity_Color, Quantity_TOC_RGB
from OCC.Display.SimpleGui import init_display

step_reader = STEPControl_Reader()
status=step_reader.ReadFile('step/ASY-0039890.stp')
step_reader.TransferRoot(1)

shapes_labels_colors = read_step_file_with_names_colors('step/ASY-0039890.stp')

my_renderer = x3dom_renderer.X3DomRenderer()
for shpt_lbl_color in shapes_labels_colors:
	my_renderer.DisplayShape(shpt_lbl_color)

my_renderer.render()`

tpaviot, 2020-04-04:
see an example at https://github.com/dblanding/kodacad

tpaviot, 2020-04-04:
Closing this issue, duplicates #805

closed 2020-04-04 by tpaviot