tpaviot/pythonocc-core #817: MakeHalfSpace Runtime Error - Only on Linux

cmartguy, 2020-04-20:
This error does not occur when running on Windows 7, but does occur when running on Linux (CentOs 7.7).  The error was:

Traceback (most recent call last):
  File "Example_Case_Fail.py", line 10, in <module>
    half_space_obj = BRepPrimAPI_MakeHalfSpace(temp_face, gp_pnt_obj)
  File "./.conda/envs/occt/lib/python3.7/site-packages/OCC/Core/BRepPrimAPI.py", line 238, in __init__
    _BRepPrimAPI.BRepPrimAPI_MakeHalfSpace_swiginit(self, _BRepPrimAPI.new_BRepPrimAPI_MakeHalfSpace(*args))
RuntimeError: Standard_ConstructionError
gp_Dir() - input vector has zero norm
wrapper details:
  * symname: new_BRepPrimAPI_MakeHalfSpace
  * wrapname: _wrap_new_BRepPrimAPI_MakeHalfSpace__SWIG_0
  * fulldecl: BRepPrimAPI_MakeHalfSpace::BRepPrimAPI_MakeHalfSpace(TopoDS_Face const &,gp_Pnt const &)

Code to reproduce this error is here.  Again it does not show problems in Windows:

from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeFace
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeHalfSpace
from OCC.Core.gp import gp_Pnt, gp_Dir, gp_Pln
    
gp_pnt_obj = gp_Pnt(0., 0., 0.)
gp_dir_obj = gp_Dir(1, 0, 0)
gp_pln_obj = gp_Pln(gp_pnt_obj, gp_dir_obj)
face_obj = BRepBuilderAPI_MakeFace(gp_pln_obj)
temp_face = face_obj.Face()
half_space_obj = BRepPrimAPI_MakeHalfSpace(temp_face, gp_pnt_obj)
temp_shape = half_space_obj.Shape()

Any help would be greatly appreciated.

tpaviot, 2020-04-20:
Thank you I will have a look. What is the pythonocc version you're using ?

cmartguy, 2020-04-20:
pythonocc-core = 7.4.0
occt = 7.4.0
OCCUtils-0.1 = dev

tpaviot, 2020-04-20:
On Windows 10, using latest pythonocc, I have the following result

```bash
Traceback (most recent call last):
  File "bug_occ.py", line 10, in <module>
    half_space_obj = BRepPrimAPI_MakeHalfSpace(temp_face, gp_pnt_obj)
  File "D:\Anaconda3\envs\7.4.0dlr-sc\lib\site-packages\OCC\Core\BRepPrimAPI.py", line 238, in __init__
    _BRepPrimAPI.BRepPrimAPI_MakeHalfSpace_swiginit(self, _BRepPrimAPI.new_BRepPrimAPI_MakeHalfSpace(*args))
RuntimeError: Standard_ConstructionError
gp_Dir() - input vector has zero norm
wrapper details:
  * symname: new_BRepPrimAPI_MakeHalfSpace
  * wrapname: _wrap_new_BRepPrimAPI_MakeHalfSpace__SWIG_0
  * fulldecl: BRepPrimAPI_MakeHalfSpace::BRepPrimAPI_MakeHalfSpace(TopoDS_Face const &,gp_Pnt const &)
```
Same as you, issue confirmed.

cmartguy, 2020-04-20:
Found the problem.  My inputs were incorrect.  The point should NOT be on the face or plane.  Adding my point and vector to make a new point solved the problem.

It is still strange that it worked on my Windows machine and then failed in Linux.

Thanks for your help.

tpaviot, 2020-04-21:
@cmartguy good, can you please share the working code. It's strange that this works on Linux and not on Windows. If conditions are not fulfilled to perform the HalfSpace, there's no reason that the same exception is not raised on Linux as well. I keep this issue opened until we find an explanation.

tpaviot, 2020-04-21:
@cmartguy The exception is also correctly raised on Linux (at least on my Ubuntu 18.04 machine)

```bash
$ python bug_occ_halfspace.py 
Traceback (most recent call last):
  File "bug_occ_halfspace.py", line 10, in <module>
    half_space_obj = BRepPrimAPI_MakeHalfSpace(temp_face, gp_pnt_obj)
  File "/home/thomas/miniconda3/envs/occt620/lib/python3.6/site-packages/OCC/Core/BRepPrimAPI.py", line 391, in __init__
    _BRepPrimAPI.BRepPrimAPI_MakeHalfSpace_swiginit(self, _BRepPrimAPI.new_BRepPrimAPI_MakeHalfSpace(*args))
RuntimeError: Standard_ConstructionErrorgp_Dir() - input vector has zero norm raised from method BRepPrimAPI_MakeHalfSpace of class BRepPrimAPI_MakeHalfSpace
```
We can close the issue.

closed 2020-04-21 by tpaviot