tpaviot/pythonocc-core #818: PythonOcc: Thicken a curve with variable radius

godjenka, 2020-04-21:
I am new to pythonOcc, I want to thicken a Bezier Curve (make a curved pipe) unevenly: for example, the resulting thickened curve at the start point of the curve should be thicker than at the endpoint of the curve. Could you let me know what is the most efficient way to do that? 

I've tried functions like BRepOffsetAPI_ThruSections, but I don't think there are useful for this purpose.

rainman110, 2020-04-21:
Use BRepOffsetAPI_MakePipeShell:
https://www.opencascade.com/doc/occt-6.9.0/refman/html/class_b_rep_offset_a_p_i___make_pipe_shell.html

tpaviot, 2020-04-22:
@godjenka Please feel free to contribute a sample code for using ```BRepOffsetAPI_MakePipeShell``` if ever you get it working. Indeed our pythonocc-demos repository misses an example related to this class.

@rainman110 Link update: https://www.opencascade.com/doc/occt-7.4.0/refman/html/class_b_rep_offset_a_p_i___make_pipe_shell.html

godjenka, 2020-04-28:
I will give it a try today and let you know sorry for the late reply.

godjenka, 2020-04-28:
I think it should work, but I m not too sure on how to use Handle_Law_Function to create the law. Is there an example that uses Handel_Law_Function for pythonocc?

godjenka, 2020-04-28:
Solved it. Will try to post an example as soon as I clean up everything.

closed 2020-04-28 by godjenka