tpaviot/pythonocc-core #819: JupyterRenderer Error with as1_pe_203.stp

andreasplesch, 2020-04-24:
When I try to render the as1_pe_203.stp model from the binderhub repo with JupyterRenderer, there is this error:

```
shp = read_step_file(os.path.join('..', 'assets', 'models', 'as1_pe_203.stp'))
my_renderer = JupyterRenderer(size=(800,700))
my_renderer.DisplayShape(shp, render_edges=True)
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-6-8fc317006c3e> in <module>
----> 1 my_renderer.DisplayShape(shp, render_edges=True)

/opt/conda/lib/python3.7/site-packages/OCC/Display/WebGl/jupyter_renderer.py in DisplayShape(self, shp, shape_color, render_edges, edge_color, edge_deflection, vertex_color, quality, transparency, opacity, topo_level, update, selectable)
    608             result = self.AddShapeToScene(shp, shape_color, render_edges,
    609                                           edge_color, vertex_color, quality,
--> 610                                           transparency, opacity)
    611             output.append(result)
    612 

/opt/conda/lib/python3.7/site-packages/OCC/Display/WebGl/jupyter_renderer.py in AddShapeToScene(self, shp, shape_color, render_edges, edge_color, vertex_color, quality, transparency, opacity)
    686         tess.Compute(compute_edges=render_edges,
    687                      mesh_quality=quality,
--> 688                      parallel=True)
    689         # get vertices and normals
    690         vertices_position = tess.GetVerticesPositionAsTuple()

RuntimeError: Standard_NoSuchObject
NCollection_List::First
wrapper details:
  * symname: ShapeTesselator_Compute
  * wrapname: _wrap_ShapeTesselator_Compute
  * fulldecl: void ShapeTesselator::Compute(bool,float,bool)
```

andreasplesch, 2020-04-26:
This is how the stp loops converted to vrml:
[image]
It does not have triangles, only edges.

tpaviot, 2020-04-27:
Issue confirmed

tpaviot, 2020-04-27:
The issue is raised from the Tesselator. Something fails when computing edges from the line:
```c++
const TopoDS_Face& aFace = TopoDS::Face(edgeMap.FindFromIndex(iEdge).First());
```
it's the first time I see this issue. It's strange, because if you split the basic Compound into individual shapes then it works. I don't see, so far, how to solve the issue, except raisins a proper exception. Maybe some kind of  ```ShapeFix``` is required before performing the tessellation.

tpaviot, 2020-04-27:
The ```TopoDS_Compound``` returned by the ```read_step_file``` function contains free edges for the as1_pe_203.stp file, thus leading to an exception during edge discretization.

tpaviot, 2020-04-27:
Fixed.

closed 2020-04-27 by tpaviot