tpaviot/pythonocc-core #820: Display shape without wireframe

godjenka, 2020-04-26:
This one must be easy to solve, but I don't seem to find the solution. I'd like to display a shape without wireframe. How can I do it?

tpaviot, 2020-04-27:
By default, shapes in the SimpleGui are displayed in shaded mode, not wireframe except if you explicitly ask for.

VictorLamoine, 2020-04-27:
I think he meant displaying the shape solid with no edges/wires displayed.

tpaviot, 2020-04-27:
ah, ok, in this case, just add this line:

```python
display.default_drawer.SetFaceBoundaryDraw(True)
```

tpaviot, 2020-04-27:
I confess this is not straightforward, I added an example at https://github.com/tpaviot/pythonocc-demos/commit/0a97f2d805adb48fde93a92291c21a02b1bb01b3

tpaviot, 2020-04-27:
Please @godjenka close the issue if ever the provided solution fixes the issue

godjenka, 2020-04-28:
Yes, this is what I meant, thanks!

closed 2020-04-28 by godjenka