tpaviot/pythonocc-core #821: Antialiasing not working in SimpleGui

VictorLamoine, 2020-04-27:
# Test code
```python
#/!/usr/bin/env python3
import os
import os.path
import sys
import argparse

from OCC.Display.SimpleGui import init_display
from OCC.Extend.DataExchange import read_step_file

if __name__ == "__main__":
    parser = argparse.ArgumentParser()
    parser.add_argument("file", help="STEP file to display")
    args = parser.parse_args()

    display, start_display, add_menu, add_function_to_menu = init_display(background_gradient_color1=[58,61,55], background_gradient_color2=[58,61,55])
    display.SetPerspectiveProjection()
    display.EnableAntiAliasing()
#    display.SetNbMsaaSample(4)
    display.Context.SetAutoActivateSelection(False)
    shp = read_step_file(os.path.join(args.file))
    display.DisplayShape(shp, update=True)
    start_display()
```

# Result
[image]

# Command line
```bash
$ python display_step.py ~/part.step 
INFO:OCC.Display.backend:backend loaded: qt-pyqt5
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt5
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
V3d_Viewer created.
AIS_InteractiveContext created.
V3d_View created
Xw_Window created.
Display3d class successfully initialized.
 ########################################
Warning: EnableAntiAliasing does not work as exce$
 **  Model Complete Check List  **
Check:1 -- Entity (n0:id) 49:#55   Type:(NAMED_UNIT,PLANE_ANGLE_UNIT,SI_UNIT)
Parameter n0.1 (dimensions) not Derived
Check:2 -- Entity (n0:id) 50:#56   Type:(NAMED_UNIT,SI_UNIT,SOLID_ANGLE_UNIT)
Parameter n0.1 (dimensions) not Derived
Check:3 -- Entity (n0:id) 239:#256   Type:(LENGTH_UNIT,NAMED_UNIT,SI_UNIT)
Parameter n0.1 (dimensions) not Derived
```

There is a warning: `Warning: EnableAntiAliasing does not work as exce$`

I can't find it in the source code (`pythonocc-core`) and setting the MSAA samples does not work either (`'Viewer3d' object has no attribute 'SetNbMsaaSample'`)

tpaviot, 2020-04-27:
It's already fixed in the master branch see commit 3c13195b81de0e4b2f94fce813598dfeb78414e5

closed 2020-04-27 by tpaviot

VictorLamoine, 2020-04-27:
Yes I have compiled `OpenCascade` and `pythonOCC` and it works fine now. Thanks
[image]
###

tpaviot, 2020-04-27:
ah, you set up a local compile/build workflow, good to know