tpaviot/pythonocc-core #822: TopAbs_Print()

ANkhan-93, 2020-04-28:
Hi all,
I am an infant when it comes to pythonOCC. I am still learning it and trying a few things here and there. 

I want to ask about the TopAbs package that contains the ShapeEnum enumeration for the different topology shapes. I was trying to use the " topabs_Print() " function.

-- | --
TopAbs_Compund | 0
TopAbs_Compsolid | 1
TopAbs_Solid | 2
TopAbs_Shell | 3
TopAbs_Face | 4
TopAbs_Wire | 5
TopAbs_Edge | 6
TopAbs_Vertex | 7
TopAbs_Shape | 8

Topabs_Print() : Prints the name of Shape <SEq> as a String on the Stream <S> and returns <S>.

So the first parameter <SEq> should be an integer from the table above and it should give me the corresponding topology name. I don't understand understand, what is the second parameter <S>. I have tried different things but this function doesn't work. It says the second function should be of type " Standar_OStream & ".

tpaviot, 2020-05-02:
The Standard_Ostream is some c++ specific stuff. What do you want to do ? there *is* a pythonic way to achieve it

ANkhan-93, 2020-05-05:
Hello, 
Thank you for your reply..
Actually I am working on my thesis, part of which includes learning about OCC. If its not too complicated, it would be nice if I could use this TopAbs_Print() to reconcile the different topology types with their corresponding enumeration.

Thanks