tpaviot/pythonocc-core #823: xml declaration encoding

andreasplesch, 2020-04-28:
I noticed that X3DExporter generates an xml declaration (the first line) with

```
encoding='utf8'
```

This is invalid xml:

https://www.w3.org/TR/REC-xml/#charencoding

It should read
```
encoding='UTF-8'
```

Looking into this, it turns out that ElementTree is responsible:

https://github.com/tpaviot/pythonocc-core/blob/7769c01333e43eefe48d693f07a5bba672dc1079/src/Display/WebGl/x3dom_renderer.py#L346

Accordind to this footnote

https://docs.python.org/3/library/xml.etree.elementtree.html#id9

I think the line should read:

```python
clean_x3d_str = ElementTree.tostring(xml_et, encoding='UTF-8').decode('utf8')
```

tpaviot, 2020-04-29:
It appears element tree is not very good at exporting xml declaration or doctype. ElementTree, when calling the ```.tostring``` method, removes the declaration and the doctype I defined. A better way should be to go to lxml but it adds another not necessary dependency. I pushed a simple fix: remove element tree, I will use it in the unittest suite only, to check that the xml is valid. I also updated the DTD version. Attached a simple box generated after this change, can you please check it's ok. If yes, please close the ticket.

[box.zip](https://github.com/tpaviot/pythonocc-core/files/4549920/box.zip)

andreasplesch, 2020-04-29:
InstantPlayer does not complain anymore about the .x3d xml, so this works.
Another option is to use ElementTree just for the X3D element, and add the declaration and doctype manually as strings. But unit testing for valid xml should suffice anyways.

closed 2020-04-29 by andreasplesch