tpaviot/pythonocc-core #824: converting from stp to x3d

GehadMlanh, 2020-05-01:
I am trying to convert my stp file to x3d file . I am using the following code 
# Python script simplified from the example script 
# core_load_step_ap203_to_x3d distributed with pythonOCC
input_file  = 'm12_2.stp'   # input STEP (AP203/AP214 file)
output_file = 'm12_2.x3d'   # output X3D file

from OCC.STEPControl import STEPControl_Reader
from OCC.IFSelect import IFSelect_RetDone, IFSelect_ItemsByEntity
from OCC.Visualization import Tesselator, atNormal
step_reader = STEPControl_Reader()
status = step_reader.ReadFile( input_file )

if status == IFSelect_RetDone:  # check status
    failsonly = False
    step_reader.PrintCheckLoad(failsonly, IFSelect_ItemsByEntity)
    step_reader.PrintCheckTransfer(failsonly, IFSelect_ItemsByEntity)
    aResShape = step_reader.Shape(1)
else:
    print("Error: can't read file.")
    sys.exit(0)

Tesselator(aResShape).ExportShapeToX3D( output_file )

it shows me an error in the :
from OCC.Visualization import Tesselator, atNormal ... it is not there ..
I have found it in 
from OCC.Tesselator import T_esselator.

but it still not working 

Any advice is highly appreaciated ....

tpaviot, 2020-05-01:
Update to the latest pythonocc release, and try the following code:
```python
input_file  = 'm12_2.stp'   # input STEP (AP203/AP214 file)
output_file = 'm12_2.x3d'   # output X3D file
from OCC.Extend.DataExchange import read_step_file
from OCC.Core.Tesselator import ShapeTesselator

aResShape = read_step_file(input_file)
tess = ShapeTesselator(aResShape)
tess.Compute()
tess.ExportShapeToX3D( output_file )
```

GehadMlanh, 2020-05-01:
Thank you for your answer ... It works . 
But I have a 3d design which consists of multiple parts. Is there a way to not show it as one solid part but as a multiple parts including colors. 

info to my application , actually I am trying to use x3d in the websites so that the users can select and unselect some parts. so if you think there is another way to do it using occ 
iI will appreciate your help 

Thank you

tpaviot, 2020-05-02:
See the demo https://github.com/tpaviot/pythonocc-demos/blob/master/examples/core_webgl_x3dom_bigfile_multipleshapes.py as well as the ```read_step_file_with_names_colors``` function at https://github.com/tpaviot/pythonocc-core/blob/master/src/Extend/DataExchange.py#L127