tpaviot/pythonocc-core #825: BRepOffsetAPI_MakePipeShell error when calling its Build method

glinskyc, 2020-05-02:
When attempting to make use of `BRepOffsetAPI_MakePipeShell`, a runtime error occurs when calling its `Build` method.

For an easy-to-reproduce example, I modified pythonocc demo file `core_topology_pipe.py` to try making use of `BRepOffsetAPI_MakePipeShell` instead of `BRepOffsetAPI_MakePipe`.

I added the following lines of code to `core_topology_pipe.py`:
```
from OCC.Core.BRepOffsetAPI import BRepOffsetAPI_MakePipeShell
from OCC.Core.TopTools import TopTools_ListOfShape

pipe = BRepOffsetAPI_MakePipeShell(path_wire)
profile_wire = BRepBuilderAPI_MakeWire(profile_edge).Wire()
pipe.Add(profile_wire)
cross_section_wires = TopTools_ListOfShape()
pipe.Simulate(2, cross_section_wires)
pipe.Build()
```

The following error is raised when `pipe.Build()` is called:
```
RuntimeError: Standard_OutOfRange
NCollection_Sequence::ChangeValue
wrapper details:
  * symname: BRepBuilderAPI_MakeShape_Build
  * wrapname: _wrap_BRepBuilderAPI_MakeShape_Build
  * fulldecl: void BRepBuilderAPI_MakeShape::Build()
```

I even checked `pipe.IsReady()` before calling `pipe.Build()`, and it returns `True`.
**As far as I can tell, this is a bug.**

For reference, see [BRepOffsetAPI_MakePipeShell](https://www.opencascade.com/doc/occt-7.0.0/refman/html/class_b_rep_offset_a_p_i___make_pipe_shell.html#a31f07509cff6d5294d6c9e8dcfbf2145) documentation.

tpaviot, 2020-05-02:
I need the full code to reproduce the issue on my machine

rainman110, 2020-05-02:
I think, the cross_section_wires have to be preallocated first before calling simulate:

cross_section_wires = TopTools_ListOfShape(1,2)

glinskyc, 2020-05-02:
@tpaviot here is the full code, which is a modified version of demo file `core_topology_pipe.py`:

```
from __future__ import print_function

from OCC.Core.gp import gp_Pnt
from OCC.Core.GeomAPI import GeomAPI_PointsToBSpline
from OCC.Core.TColgp import TColgp_Array1OfPnt
from OCC.Core.BRepBuilderAPI import BRepBuilderAPI_MakeEdge, BRepBuilderAPI_MakeWire
from OCC.Core.BRepOffsetAPI import BRepOffsetAPI_MakePipe

from OCC.Display.SimpleGui import init_display
display, start_display, add_menu, add_function_to_menu = init_display()

def pipe():
    # the bspline path, must be a wire
    array2 = TColgp_Array1OfPnt(1, 3)
    array2.SetValue(1, gp_Pnt(0, 0, 0))
    array2.SetValue(2, gp_Pnt(0, 1, 2))
    array2.SetValue(3, gp_Pnt(0, 2, 3))
    bspline2 = GeomAPI_PointsToBSpline(array2).Curve()
    path_edge = BRepBuilderAPI_MakeEdge(bspline2).Edge()
    path_wire = BRepBuilderAPI_MakeWire(path_edge).Wire()

    # the bspline profile. Profile mist be a wire
    array = TColgp_Array1OfPnt(1, 5)
    array.SetValue(1, gp_Pnt(0, 0, 0))
    array.SetValue(2, gp_Pnt(1, 2, 0))
    array.SetValue(3, gp_Pnt(2, 3, 0))
    array.SetValue(4, gp_Pnt(4, 3, 0))
    array.SetValue(5, gp_Pnt(5, 5, 0))
    bspline = GeomAPI_PointsToBSpline(array).Curve()
    profile_edge = BRepBuilderAPI_MakeEdge(bspline).Edge()

    # pipe
    # pipe = BRepOffsetAPI_MakePipe(path_wire, profile_edge).Shape()

    # -------------------------------------------------------------------------
    # TODO trying BRepOffsetAPI_MakePipeShell instead of BRepOffsetAPI_MakePipe
    # -------------------------------------------------------------------------
    from OCC.Core.BRepOffsetAPI import BRepOffsetAPI_MakePipeShell
    from OCC.Core.TopTools import TopTools_ListOfShape

    pipe = BRepOffsetAPI_MakePipeShell(path_wire)
    profile_wire = BRepBuilderAPI_MakeWire(profile_edge).Wire()
    pipe.Add(profile_wire)
    cross_section_wires = TopTools_ListOfShape()
    pipe.Simulate(2, cross_section_wires)
    pipe.IsReady()
    pipe.Build()

    display.DisplayShape(profile_edge, update=False)
    display.DisplayShape(path_wire, update=False)
    display.DisplayShape(pipe, update=True)

if __name__ == '__main__':
    pipe()
    start_display()
```

glinskyc, 2020-05-02:
@rainman110 the suggested preallocation did not work unfortunately.

When trying `cross_section_wires = TopTools_ListOfShape(1, 2)`, this error is raised:
```
TypeError: new_TopTools_ListOfShape expected at most 1 arguments, got 2
Additional information:
Wrong number or type of arguments for overloaded function 'new_TopTools_ListOfShape'.
  Possible C/C++ prototypes are:
    NCollection_List< TopoDS_Shape >::NCollection_List()
    NCollection_List< TopoDS_Shape >::NCollection_List(opencascade::handle< NCollection_BaseAllocator > const &)
    NCollection_List< TopoDS_Shape >::NCollection_List(NCollection_List< TopoDS_Shape > const &)
```

When trying `cross_section_wires = TopTools_ListOfShape(2)`, this error is raised:
```
TypeError: in method 'new_TopTools_ListOfShape', argument 1 of type 'NCollection_List< TopoDS_Shape > const &'
Additional information:
Wrong number or type of arguments for overloaded function 'new_TopTools_ListOfShape'.
  Possible C/C++ prototypes are:
    NCollection_List< TopoDS_Shape >::NCollection_List()
    NCollection_List< TopoDS_Shape >::NCollection_List(opencascade::handle< NCollection_BaseAllocator > const &)
    NCollection_List< TopoDS_Shape >::NCollection_List(NCollection_List< TopoDS_Shape > const &)
```

So it looks like it expects an argument of type `NCollection_List< TopoDS_Shape >`.  I'm not sure exactly what that should look like.  Any suggestions?

glinskyc, 2020-05-02:
I believe I've identified the "issue".  I incorrectly thought that `pipe.Simulate()` needed to be called prior to `pipe.Build()`, but that seems to *not* be the case.  In the example I provided, just remove the `cross_section_wires` and `Simulate` call, and simply call `pipe.Build()` to construct the swept surface.  I think you can even skip directly to calling `pipe.Shape()` since I believe that calls `Build` anyway.

It still seems odd to me that calling `Simulate` effectively breaks that ability to call `Build`, but I guess that's how it works.  I don't know if that will be problematic for certain use cases or not.

glinskyc, 2020-05-02:
Feel free to close this.  

As a final note, it would be nice to have a demo file to illustrate capabilities of `BRepOffsetAPI_MakePipeShell` that set it apart from `BRepOffsetAPI_MakePipe`.

tpaviot, 2020-05-04:
You're right, this discussion/issue has to be captured into a new demo file. I will close this ticket as soon as I tested your solution/wrote the demo.

PR #896 (2020-10-03): Review/typedef aliases
Fix #825
closed 2020-10-03 by tpaviot