tpaviot/pythonocc-core #828: core_ocaf_unittest.TestOCAF: Segmentation fault (core dumped)

roipoussiere, 2020-05-08:
I get a segfault when running the test suite (`python3 ./run_tests.py`) after successfully building python-occ from sources.

## Build options

### environment variables

- `LD_LIBRARY_PATH: /app/lib`

### config options

- `-DOCE_INCLUDE_PATH:PATH=/app/include/opencascade`
- `-DOCE_LIB_PATH:PATH=/app/lib`
- `-DPYTHONOCC_INSTALL_DIRECTORY:PATH=/app/lib/python3.7/site-packages/OCC`

## Stacktrace

```
test_write_step_file (core_ocaf_unittest.TestOCAF)
Exports a colored box into a STEP file ... 
*******************************************************************
******        Statistics on Transfer (Write)                 ******

*******************************************************************
******        Transfer Mode = 0  I.E.  As Is       ******
******        Transferring Shape, ShapeType = 2                      ******
Segmentation fault (core dumped)
```

## Installed dependencies

- glu 9.0.1
- freeglut 3.2.1
- tcl 8.6.10
- tk 8.6.10
- libXmu 1.1.3
- opencascade 7.4.0
- swig 4.0.1
- pythonocc 7.4.0

## Platform

- Linux distribution: Manjaro

tpaviot, 2020-06-16:
ok, that's something I noticed on osx as well. I have no clue so far where it comes from

karl-nilsson, 2020-11-05:
I ran into the same crash.
When saving a STEP AP203 file without specifying the author/organization, OCCT uses the computer's IP address as the default value. This function (OSD_Host::InternetAddress) segfaults, crashing the program.
I've notified OCCT upstream