tpaviot/pythonocc-core #829: label.FindAttribute needs handle

andreasplesch, 2020-05-09:
https://www.opencascade.com/doc/occt-7.4.0/overview/html/occt_user_guides__ocaf.html#occt_ocaf_3_5_1

explains how to use FindAttribute on TDF_Label.

It needs to be passed a handle to a TDF_Attribute. I saw

#720

but the problem here is that TDF_Attribute does not have a constructor, TDF_Attribute().

https://www.opencascade.com/doc/occt-7.4.0/refman/html/class_t_d_f___attribute.html

Probably I am missing something obvious. How do use TDF_Label.FindAttribute() with python ?

The python doc at  https://raw.githack.com/tpaviot/pythonocc-documentation/master/api_doc/7.4.0beta/OCC.Core.TDF.html?highlight=findattribute#OCC.Core.TDF.TDF_Label.FindAttribute
still refers to ```opencascade::handle<TDF_Attribute> &```.

andreasplesch, 2020-05-09:
For attributes which have a Set() method one can iterate over all attributes and then use:

```python
attrIt = TDF_AttributeIterator(child)
    ## Find reference
    while (attrIt.More()):
        attr = attrIt.Value()
        if (attr.ID() == wantedAttrGUID):
            foundAttr = concreteTDL_Attribute.Set(label, wantedAttrGUID)
```
Most have .Set but not all.

closed 2020-05-09 by andreasplesch

cz-source, 2022-10-20:
>
I have the same problem. Have you solved it?

nochat1205, 2023-03-15:
I solve fit with follow codes, but I think it would have Potential errors which I don't sense.
`
set_TopAttr = (
    TNaming_NamedShape.GetID(),
    XCAFDoc_Location.GetID()
)
def ShallowCopy(dest, src):
    dest.__class__ = src.__class__ # 
    dest.__dict__ = src.__dict__

    # __annotations__
    # __doc__
    # __module__

def FindAttribute(label:TDF_Label, GUID:Standard_GUID, attribute: TDF_Attribute):
    it_attr = TDF_AttributeIterator(label)
    while it_attr.More():
        # TODO: may be all object associated TopoDS should use ShallowCopy 
        if it_attr.Value().ID() == GUID:
            if GUID == TNaming_NamedShape.GetID():
                ShallowCopy(attribute, TNaming_NamedShape.DownCast(it_attr.Value() ) )
            elif GUID == XCAFDoc_Location.GetID():
                ShallowCopy(attribute, XCAFDoc_Location.DownCast(it_attr.Value() ) )
            else:
                attribute.Restore(it_attr.Value())
            return True 
        it_attr.Next()

    attribute = None
    return False

# fix
TDF_Label.FindAttribute = FindAttribute
`

nochat1205, 2023-03-15:
I solve it with the follows code . but tthink it  would have potential problem which I couldn't sense.
have the better solve?
```
set_TopAttr = (
    TNaming_NamedShape.GetID(),
    XCAFDoc_Location.GetID()
)
def ShallowCopy(dest, src):
    dest.__class__ = src.__class__ # 
    dest.__dict__ = src.__dict__

    # __annotations__
    # __doc__
    # __module__

def FindAttribute(label:TDF_Label, GUID:Standard_GUID, attribute: TDF_Attribute):
    it_attr = TDF_AttributeIterator(label)
    while it_attr.More():
        # TODO: may be all object associated TopoDS should use ShallowCopy 
        if it_attr.Value().ID() == GUID:
            if GUID == TNaming_NamedShape.GetID():
                ShallowCopy(attribute, TNaming_NamedShape.DownCast(it_attr.Value() ) )
            elif GUID == XCAFDoc_Location.GetID():
                ShallowCopy(attribute, XCAFDoc_Location.DownCast(it_attr.Value() ) )
            else:
                attribute.Restore(it_attr.Value())
            return True 
        it_attr.Next()

    attribute = None
    return False

# fix
TDF_Label.FindAttribute = FindAttribute

```

aruig666, 2024-05-24:
I have the same problem. Have you solved it?