tpaviot/pythonocc-core #830: StlAPI_Writer doesn't seem to work

stcmeh, 2020-05-10:
Hi there,

I'm new to pythonocc and I've managed to install 7.4.0

As I'm new I want to output stuff to openSCAD using STL so I can also view them there.

The following code seems simple enough and was copied from a website somewhere. I may have modified it for 7.4.0 by adding the .Core stuff.

When I run it nothing happens. The call to Write returns False.

I've tried it with a full path to the destination file - still doesn't work. I removed the path from this example for simplification.

Am I doing something wrong?

Simon

```
from OCC.Core.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Core.StlAPI import StlAPI_Writer
box_shape = BRepPrimAPI_MakeBox(50,50,50).Shape()

stl_writer = StlAPI_Writer()
stl_writer.Write(box_shape,"box.stl")
```

stcmeh, 2020-05-10:
Got it - sorry. As per your example, you need to mesh it first with BRepMesh_IncrementalMesh

closed 2020-05-10 by stcmeh

wu-yi-hong, 2023-10-27:
BRepMesh_ IncrementalMesh will cause BRepAlgoAPI_ Cut is invalid

tpaviot, 2023-10-27:
Can you please post a code snippet?